COVESA/vehicle_signal_specification #516: Introduction of Domain agnostic information model

erikbosch, 2022-12-12:
This issue is created for managing follow-up discussions on the [presentation](https://docs.google.com/presentation/d/1PHijKSHKbP2ZjmWJ4qm4GKuGID5B0TNn/edit?usp=share_link&ouid=102950097033494215951&rtpof=true&sd=true) from @UlfBj on how VSS can be used to support multiple trees.

In short:

- Split VSS-project to two projects - one managing definition of syntax and tooling and one managing the actual catalog
- New format (HIM-file) to specify trees supported 

The topic was discussed at VSS meeting 2022-12-06. The proposal was considered as interesting as participants and shall be further discussed at a later stage, but it is not considered urgent. Could be interesting to sync with VSSo. Possible topic for discussion at next AMM

erikbosch, 2022-12-12:
Next time we discuss this we could also discuss if we would need (in long term) possibility to cross-reference trees. A hypothetical example could be that VSS specifies a signal `Vehicle. MountedTollTransponders` which refers to an array of instances of another tree. This seems to be as a use-case that could be related to the proposal.

```
Vehicle. MountedTollTransponders:
  datatype: org.someorg.Transponder[]
  type: reference
  description: Reference to mounted transponders, if any
```

UlfBj, 2022-12-13:
I think that cross-referencing other trees should be limited to data types, which is the way we already have decided on in our complex datatype discussion for the VSS taxonomy. It seems that model is applicable also to your example above, that the cross referencing is to a data type definition associated to another domain?
The "type: reference" expression above looks strange to me. 

Part of the idea behind this "domain agnostic information model" is to avoid having trees that are a mix of multiple domains. So instead of a cross referencing as proposed above, it could be possible to use the "proxy" concept that is mentioned in the presentation. According to the proposed model, the HIM configuration file would then contain a declaration of the "org.someorg" Data taxonomy, with "type:proxy". This declares that the server managing this "taxonomy forest" knows how to access the MountedTollTransponders data (requires the full path into that tree), and is able to provide it in a response to a request. This keeps the VSS tree from being "bloated" with data from other domains, while still enabling access to it. 
The HIM configuration for this domain could alternatively say "type:taxonomy", in which case the server would have a copy of the entire tree, allowing it to use other metadata to verify the client request, etc.

erikbosch, 2022-12-13:
I am not sure that I fully understand your comment, I think we need better examples on how the "proxy" concept in your proposal is supposed to be used.

Assume a signal in VSS vehicle tree maintains the list of current passengers. COVESA does not maintain the content of the passenger tree . An app wants to check if any passenger has birthday today, something like:

```
for (passenger in vehicle->getPassengers()){
  if (passenger->getBirthday() == today) {
    sing_happy_birthday(passenger->getName());
  }
}
```

If the VSS tree actually want to contain information on all passengers then it would be easy to add a signal that use the type/struct passenger from some other taxonomy. The alternative is to have some form of reference/pointer solution. If the example above would be realized with VISS and HTTPS the reference you would get for each passenger would likely be a URL, either on the same server or to a totally different server. 

But I guess all of this can be clarified when we start detailing use-cases for your proposal

UlfBj, 2022-12-13:
It would have to be the responsibility of the vehicle to obtain the identities of the passengers as I see it. So this information would reside in the Vehicle tree, but it would only be the data representing the identity (Vehicle.Cabin.PassengerIdentity). The HIM configuration then declares a "proxy" tree that contains personal data such as birthday, home address, etc. This tree could then be available as a cloud service, and the server would know how to use that service to get the birthday info from the identities in the vehicle tree. The HIM path to obtain the birthday could look like PassengerIdentity.Birthday. 

When thinking more on this, I think it might be a cleaner design to have a Passenger domain that defines passenger data: identity, birthday, home address, shoe size, etc. Then this tree could be defined in the HIM config file as "taxonomy", i. e. the server manages a copy of the tree, where the identities are locally stored, while the data base containing personal data is a cloud service that the vehicle server uses to get the personal data that is requested. So the "proxy" type would not be used for this use case. 

I think the EV charging use case is a good example where the "proxy" type could be used.

SebastianSchildt, 2022-12-20:
Regarding the slides, my 2 ct:

I _agree_ a better split between "model" and "catalogue" would probably be good, just speaking of repos, I fell

 - all the `docs` would need to be removed from this repo, and they would belong together with the tools and then be named something else
 - The repo VSS should only have "spec" (the standard catalogue)

That would also make it more obvious for projects like EV charging, how:  to proceed: Nice standard, lets build a model based on that

I _disagree_ with a super generic term/scope such as the proposed HIM. I think there are so many standards in so many domains, and we should not compete with them. Even if the argument is, _we_ do not do that, but others could, it is still a matter of "marketing"/"branding". "Hierarchical Information Model" is "too big" and just as a name claims everything and nothing. At least "Vehicle Signal Specification", (not a very good name for the ruleset either , I agree), gives you immediate recognition where it might come from/is used. Complaining aside, I do not have really good suggestion.

I _agree_ that as a consequence in systems/deployments you may have several trees (actually you already have it today, when dealing with multiple Vehicles), I do however _disagree_ that somehow define "linking" it on the YAML spec level as we have it today  is a good idea. It would end up halfway reinventing RDF or JSON-LD maybe, and in a specific application (e.g. something that deals with vehicles and charging infrastructure) it will likely be "obvious" what to do in a tech stack, without a "model" describing "the system has both", helping much.

I am open to be swayed either way on all these points 😁

UlfBj, 2023-01-02:
Choosing a domain agnostic name is meant to signal that this model is not exclusive to the automotive domain. That does not imply an ambition to compete with other models, but merely to signal that it is open for being used in other domains also. If "neighboring" domains would do so, I think it might become a valuable synergy effect. 

In a scenario where it is obvious, and static, which trees are part of the forest being used, I might agree that having a configuration file is unnecessary. However, I believe it is very likely that the set of trees will be dynamic and change over time. If so, the lack of a configuration file would be a serious restriction. Having a standardized configuration file also enables the use case of "forest discovery", that a connecting client can be informed about the set of trees the server manages. Excluding a configuration file will not strengthen the model from an interoperability point of view.

erikbosch, 2023-01-31:
Meeting notes:

- Good topic for next AMM.

closed 2024-11-14 by erikbosch