COVESA/vehicle_signal_specification #517: HMI TirePressureUnit - allowed values ‘PSI’, ‘KPA’, ’BAR’ - Quotes differs from other enum - Is this intentional

DSE8COB, 2022-12-16:
HMI.TemperatureUnit:
  datatype: string
  type: actuator
  allowed: ['C', 'F']
  description: Temperature unit used in the current HMI

HMI.TirePressureUnit:
  datatype: string
  type: actuator
  allowed: [‘PSI’, ‘KPA’, ’BAR’]
  description: Tire pressure unit used in the current HMI 

---------------------------
HMI TirePressureUnit - allowed values ‘PSI’, ‘KPA’, ’BAR’ - Quotes differs from other enum - Is this intentional

allowed: [‘PSI’, ‘KPA’, ’BAR’] - Quotes used is ’ but it should be '

DSE8COB, 2022-12-16:
Please go to the file https://github.com/COVESA/vehicle_signal_specification/blob/master/spec/Cabin/Infotainment.vspec

and see in the file then you could feel the difference in quotes. -> Am not able to upload an image

erikbosch, 2022-12-16:
Good catch, I created a PR for it in #518

PR #518 (2022-12-21): Fix quotes for TirePressureUnit
Fixes #517
closed 2022-12-21 by erikbosch