COVESA/vehicle_signal_specification #519: #include directive and YAML

gadams999, 2022-12-17:
Starting to build a Node.js parser to create a signal catalog and ran past the `#include` directive. This is challenging in that normal YAML parsing libraries do not parse comment fields.

I'm curious 1/how it was selected, 2/thoughts on a specification change (or addition) to incorporate the `include` keyword into the schema, and 3/anyone know of a good YAML parser in javascript that can read the comment field? Looks like `js-yaml` doesn't parse at all.

erikbosch, 2023-01-02:
I do not know the background on why `#include` was selected originally. I guess any proposal that easily could be integrated by existing tooling could be considered. There is by the way a tool `vspec2yaml` in [vss-tools](https://github.com/COVESA/vss-tools) that flattens (expands `#include`) and expand instances and create a "flat" Yaml representation. do not know if first using that helps your use case.

erikbosch, 2023-02-02:
I came to to think about this issue when working on vss-tools code. There is a method `yamilify_includes`, currently at https://github.com/COVESA/vss-tools/blob/master/vspec/__init__.py#L665 that contains the following comment. So I assume it would be quite easy, if wanted, to change to that Yaml syntax if we would like to.

```

#
# Convert
#   #include door.vspec, body.door.front.left
# to
#   - $include$:
#     file: door.vspec
#     prefix: body.door.front.left
#
# This yaml version of the include directive will
# then be further processed to actually include
# the given file.
#
```

closed 2024-11-14 by erikbosch