COVESA/vehicle_signal_specification #524: Solar panel specifications

CorvetteCole, 2023-01-18:
I am planning on adding some specifications for solar panels for a car I am working on. I suspect this would fit under the powertrain module, although some things are unclear. The car will technically be charging and discharging at the same time with solar, which makes some things unclear.

Looking for input! Notably our solar panels will report whether they are enabled and basic electrical specifications regarding how much power they are outputting

erikbosch, 2023-01-18:
Some quick thoughts:

- I think signals representing solar panels could fit into VSS, as long as they describe a solution that possibly could be reused by other OEMs. I.e. describing a generic solar panel solution rather than your specific solution.
- As you mention powertrain I assume it is only relevant for electrical cars, to charge the traction battery, i.e. the signals are not intended to cover scenarios where you use solar panels to charge the supply battery in e.g. a mobile home.
- We have already signals describing net input/output over the battery (e.g. `Vehicle.Powertrain.TractionBattery.CurrentPower`) and we have signals specified to charging in `Vehicle.Powertrain.TractionBattery.Charging`. We would possibly need to change description of `Charging` branch to state that charging by solar panel is not covered by that branch.
- I could imagine having either a branch  `Vehicle.Powertrain.TractionBattery.SolarPanel` or a branch `Vehicle.Powertrain.SolarPanel` that describes characteristics of the solar panel, where signals with current voltage and current (or power) from the solar panel resides. I.e. describing what the solar panel itself delivers, if you want to know the net result on the battery you need to look at `Vehicle.Powertrain.TractionBattery.CurrentPower`. 

What you possibly could do is start analyzing what signals you would need and how a possible extension to COVESA could look like. If you just want early feedback you could do a dummy pull request within your own fork and mention it in a comment in this issue and ask for feedback. We could also if needed bring it up for discussion in one of the weekly calls (16.00 CET). When stable an official PR could be created.

CorvetteCole, 2023-01-18:
Sounds good, I'll do a little work and open a draft PR for feedback. Some vehicles also have solar panels for charging the low voltage battery, although I think newly released cars do focus on charging the traction battery (e.g. Aptera, Lightyear 0, Tesla Cybertruck).

Thanks for the detailed next steps!

CorvetteCole, 2023-01-18:
Although the charging state wouldn't cover being charged by an onboard solar panel, I wonder if the solar panel should still count for AccumulatedChargedEnergy?

I think it may also be an option to add something like a "SOLAR" or "TRICKLE" to Charging.Mode. Just a thought

erikbosch, 2023-01-18:
Concerning `AccumulatedChargedEnergy` and similar signals, I guess it will need to be decided on case to case basis. As I interpret that signal today it just consider the net flow in to the battery, regardless of where the energy comes from.

closed 2024-11-14 by erikbosch