COVESA/vehicle_signal_specification #532: Semantics of Vehicle.Speed

sophokles73, 2023-02-10:
It seems like some domains like fleet management make a distinction between what some call *wheel-based* speed vs. the value that is displayed on the tachometer.

The vspec file's definition of the *Vehicle.Speed* data entry is quite unspecific as to the source of the value.
Can you shed some light on this?

erikbosch, 2023-02-10:
The `Vehicle.Speed` in VSS as of today does not specify how the value is calculated. So it might be based on wheel speed, but could as well be based on GNSS, radar or some other technology. It also does not specify preferred values in certain corner cases, like if you are stuck in a mud pool (with wheels rotating) or sliding down an icy hill (with wheels moving slower than vehicle or not at all). For wheel speed we have today speed for individual wheels like `Vehicle.Chassis.Axle.Row1.Wheel.Left.Speed` but no specific signal for average wheel speed. If needed we could improve the description of the `Vehicle.Speed`. It has sometimes been briefly defined and then most have seemed to agree that it preferably should represent true speed in "forward direction" relative to the surface the vehicle is standing on, but that actual implementations may not follow that for all corner cases.  

Concerning what is shown on HMI/Tachometer - as of today VSS does not have any signal for speed shown, like [PERF_VEHICLE_SPEED_DISPLAY](https://android.googlesource.com/platform/hardware/interfaces/+/master/automotive/vehicle/2.0/types.hal) in Android VHAL. That could be added/proposed if needed.

sophokles73, 2023-02-13:
Thanks for the thorough explanation. IMHO all of what you said would be very helpful in the description (or comment?) for *Vehicle.Speed*.

I believe that for commercial vehicles like trucks and buses/coaches the vehicle's current speed is defined in a more formal way simply because it needs to be recorded on a tachograph. Maybe that would be helpful as well, though I have no access to that definition.

> So it might be based on wheel speed, but could as well be based on GNSS, radar or some other technology.

Given that there already is a *Vehicle.CurrentLocation* branch, it might make sense to add *Speed* to that branch which (obviously) would contain the GNSS based/calculated speed of the vehicle.

closed 2024-11-14 by erikbosch