COVESA/vehicle_signal_specification #534: Sub sequent uses of the #include command are not supported

jdacoello, 2023-02-10:
The command `#include` only works in one level. Sub sequent uses of it are not considered when the final `vspec` files are produced. Is this done by design, or shall we improve that command to be used multiple times in the hierarchy?

erikbosch, 2023-02-10:
It shall work. 

Are you referring to cases when there are multiple includes after each other like:

```
#include Vehicle/VersionVSS.vspec Vehicle
#include Vehicle/VehicleIdentification.vspec Vehicle
```
Or when file A include file B, and file B include include file C?

The latter is quite commonly used today, like `VehicleSignalSpecification.vspec`includes `Body.vspec`, which in turn includes `StaticLights.vspec`

if you have an example I can troubleshoot.

(Note - there were a bug #145 that I fixed yesterday - previously include on last line was ignored if there were no newline)

jdacoello, 2023-02-14:
I followed the example you mentioned and it worked now. Apparently, I was using that command wrongly. I will close this issue.

closed 2023-02-14 by jdacoello