COVESA/vehicle_signal_specification #539: Comments are not shown in the resulting specification

jdacoello, 2023-02-14:
I realized that the comments added in the `vspec` files are not part of the resulting specification files produced by the `vss-tools`. Is this a design decision, or do the tools need an update? See an example below:

### Model
```
PerceivedAmbientLight:
  datatype: uint8
  type: sensor
  unit: percent
  min: 0
  max: 100
  description: The percentage of ambient light that is measured (e.g., by a sensor) inside the cabin. 0 = No ambient light. 100 = Full brightness.
  comment: See also Vehicle.Cabin.Lights.AmbientLight, which is used for acting on the decorative coloured lights available in the cabin.
```

### Resulting specification
```
Vehicle.Cabin.Lights.PerceivedAmbientLight:
  datatype: uint8
  description: The percentage of ambient light that is measured (e.g., by a sensor) inside the cabin. 0 = No ambient light. 100 = Full brightness.
  max: 100
  min: 0
  type: sensor
  unit: percent
  uuid: fc0b26067ecc589291c3e5264551bfb4
```

erikbosch, 2023-02-14:
Interesting, it shall generally work, see test cases like in https://github.com/COVESA/vss-tools/blob/master/tests/vspec/test_comment/expected.yaml (I assume it is vss2yaml you use?)

erikbosch, 2023-02-14:
Is it with the code in #540 you experienced it, if so I can try to trouleshoot it. Which vss-tools version did you use? There are quite a lot ongoing right now for vss-tools.

jdacoello, 2023-02-14:
Yes, I experience that issue with the code in #540. I checked, and you are right. The comments appear in the JSON for example, but not in the YAML file.

erikbosch, 2023-02-15:
Hi!

I think I have found the problem. There were previously a bug that comments were not included in Yaml, it was fixed by [this PR](https://github.com/COVESA/vss-tools/pull/184) around 6 months ago. Your PR #540 is however based on a quite old VSS-version, Github reports that your branch is 5 commits ahead and 56 commits behind COVESA master. Maybe you started from the [master-branch in your own fork](https://github.com/jdacoello/vehicle_signal_specification/tree/master), which is 56 commits behind. If you then use the default vss-tools (the one included as submodule) then you get an old vss-tools, before the fix was merged. I suggest rebasing your PR on latest COVESA master and check if the problem has been solved.

[image]

jdacoello, 2023-02-21:
I was indeed using an older version of the tools. I thought I did pull the latest changes before working. But apparently that was not the case. I will close this issue because it was already solved.

closed 2023-02-21 by jdacoello