COVESA/vehicle_signal_specification #544: Wrong types defined in OBD.vspec

jdacoello, 2023-02-17:
Some corrections are needed for the types defined in [OBD.vspec](https://github.com/COVESA/vehicle_signal_specification/blob/2b24d0d9ac6b2d357e1d97ca3064cc2ca5bfa089/spec/OBD/OBD.vspec).

For example:

```
FuelType:
  datatype: string
  type: sensor (it should be attribute)
  description: PID 51 - Fuel type
```

```
Status.IgnitionType:
  datatype: string
  type: sensor (it should be attribute)
  allowed: ['SPARK', 'COMPRESSION']
  description: Type of the ignition for ICE - spark = spark plug ignition, compression = self-igniting (Diesel engines)
```

BTW, what is the take on OBD data described in VSS? Is this actually used?

erikbosch, 2023-02-17:
Have not heard about anyone actually using them. I have done some corrections in the past to solve errors. Feel free to create a PR for the errors above.

erikbosch, 2023-09-11:
Merged by #587