COVESA/vehicle_signal_specification #545: Limit the number of characters with string datatype

jdacoello, 2023-02-21:
### Discussed in https://github.com/COVESA/vehicle_signal_specification/discussions/533

<div type='discussions-op-text'>

<sup>Originally posted by **jdacoello** February 10, 2023</sup>
Is it possible to define the number of valid characters when using string datatype? One example would be to set light colours using hexadecimal color code, where 7 characters are expected.

* `#ABCDEF`

If not, I can create an issue to work on that.</div>

Another example where this function can help is the `VIN` number:

```
VehicleIdentification.VIN:
  datatype: string
  type: attribute
  description: 17-character Vehicle Identification Number (VIN) as defined by ISO 3779.
```

In practice, how do you enforce the use of 17 characters? If this characteristic is only mentioned in the description, one must have a custom function that does the job. In contrast, if a field in VSS explains the constraints, the user can refer to the described constraints.

erikbosch, 2023-02-21:
There is no mechanism for that today. We have a similar topic for strings where the description says it shall be time formatted according to ISO 8601.  One thought I had at a time was to "invent" special units, like `unit: VIN` , `unit: color` and describe the expected formatting as a comment in the [unit file](https://github.com/COVESA/vehicle_signal_specification/blob/master/spec/units.yaml). Then one could even possibly add a regexp to the unit definition file which tooling/servers/client possibly could use for validation.

jdacoello, 2023-02-21:
> There is no mechanism for that today. We have a similar topic for strings where the description says it shall be time formatted according to ISO 8601. One thought I had at a time was to "invent" [...]

I would not use the `unit` for something that is not a unit. However, the approach taken for handling units could be reused with a new field. So, instead of defining that as a unit, we can do it as `format` (or some similar keyword).

erikbosch, 2024-02-08:
I came up with a possible but maybe somewhat ugly idea for the string datatype. We have already `min` and `max` attributes, we typically use them today just for int/float values. We could possibly give them a different semantic meaning for signals of type `string` and `string[]` and let them indicate min/max length for the string (excluding trailing null).