COVESA/vehicle_signal_specification #567: Last Touch by User to Infotainment

UmangSharmaMobis, 2023-03-27:
In phone as we see, phone goes to sleep mode if not touch for long time, same use case can be use to save infotainment battery by using last touch VSS Signal, that can help OEMS.
so i am proposing a signal which having information regarding last touch in power optimization's branch.

erikbosch, 2023-03-28:
Feel free to create a PR proposal. It could possibly be added to the [Infotainment](https://github.com/COVESA/vehicle_signal_specification/blob/master/spec/Cabin/Infotainment.vspec) branch, possibly something like below. 

I see at least two topics to consider:

- Does it only concern touching screen, or shall it also consider other events like SMS received (if connect to a phone using bluetooth) or volume adjusted by knob/button on steering wheel, where it is quite common that something is shown on screen?
- Shall it be a string ISO 8601 timestamp or something else?

```
LastHMIAction:
  datatype: string
  type: sensor
  description: Time for next charging-related action, formatted according to ISO 8601 with UTC time zone.
```

UmangSharmaMobis, 2023-03-31:
1. yes, it is  also consider for other event also like SMS received, any alert, information display popup comes with knob and other user actions.
2. yes,  it is string ISO 8601 timestamp.

erikbosch, 2023-04-04:
Meeting notes: Please discuss, to be brought up next week

UmangSharmaMobis, 2023-04-17:
https://github.com/COVESA/vehicle_signal_specification/pull/577

UmangSharmaMobis, 2024-01-24:
signal has been added #577

closed 2024-01-24 by UmangSharmaMobis