COVESA/vehicle_signal_specification #572: Display Timer off for Infotainment

UmangSharmaMobis, 2023-04-05:
In phone as we see, phone goes to sleep mode if not touch for  some desire time, same use case can be use to save infotainment battery by  defining desire time.
so i am proposing a signal which having information regarding desire time by which infotainment might goes to display sleep mode.

erikbosch, 2023-04-06:
I suggest that we briefly discuss your issues on the next VSS meeting. I see no specific problems if we would add a `Vehicle.Cabin.Infotainment.HMI.DisplayOffTime` or similar to be used in combination with a `Vehicle.Cabin.Infotainment.HMI.LastEventTime` or similar as proposed in the other issue. The question is if we also need a signal `IsScreenAlwaysOn` that can be set by applications to prevent the screen from going black, for example if using the back camera, important errors/messages are shown or similar.

UmangSharmaMobis, 2023-04-17:
please find pull request https://github.com/COVESA/vehicle_signal_specification/pull/577

erikbosch, 2023-05-04:
Merged by #585 I think it can be closed

closed 2023-05-05 by UmangSharmaMobis