COVESA/vehicle_signal_specification #586: DMS signals now in two locations?

nickrbb, 2023-05-02:
In PR #573 the branch Vehicle.ADAS.DMS was added, which is described as "Driver Monitoring System signals". This branch contains fairly high-level signals, indicating such things as enablement, warning, and error. However, there are other DMS-related signals that provide much more detailed indications under the branch of Vehicle.Driver. These include the following:
- Vehicle.Driver.DistractionLevel
- Vehicle.Driver.IsEyesOnRoad
- Vehicle.Driver.IsHandsOnWheel
- Vehicle.Driver.AttentiveProbability
- Vehicle.Driver.FatigueLevel
- Vehicle.Driver.HeartRate

Should the above signals be moved under the new Vehicle.ADAS.DMS branch, so that all driver monitoring-related signals appear in one location?

JochenKienzle, 2023-05-02:
I also have had questions and discussions about the problem behind this topic.

The grouping of signals can be done in different ways (logical, technical, functional, physical, ...) and the best fit depends on the related use cases which are likely different. 
In this example, driver is more like a physical grouping (a driver "owns" / "defines" these signals) but ADAS is more a functional/technical grouping (because these signals mostly used in ADAS context). Since VSS is a more generic definition, I see the need to have "**multiple views**" on the same 'base' signal-tree. 

So, my suggestion for this would be:

- Long-term: define base-grouping type & related rules for VSS
- find a single owner of a signal using this base-grouping (e.g. Driver)
- allow references from other branches/views to this signal (e.g. ADAS)
- Long-term & if needed: provide different views and/or only mechanism to define them (e.g. functional, physical, ...)

What should be considered:

- only one owner, otherwise there is the risk of out-of-sync signals
- avoid that specific views/groups are used for exchange / sharing data (therefore a base-grouping or a defined set of views shall be used).

pmundt, 2023-05-31:
My apologies for jumping in to this late, I did not see the notification on the GitHub issue and have missed the last couple of VSS calls.

To me the two branches are focusing on different items:

- `Vehicle.ADAS.DMS` is concerned with the state of the DMS system itself (e.g. whether it's enabled, in an error state, etc.)
- `Vehicle.Driver` is concerned with the results of the detection from the active DMS system as they apply to the driver

Having some sort of mechanism for linkage or grouping would be nice to have (e.g. the Driver signals are grouped together with the DMS signals, but are only active when the DMS system is, and can't be relied upon when the system is disabled or in an error state). Perhaps we need some sort of mechanism to map signal correlation and provide a mechanism for de-asserting/invalidating dependent signals when the parent goes high/low?

Furthermore, I am currently working on a proposal for driver alerting signals based on the detection results, which I'm also not entirely sure where to put yet, but am happy to get feedback on once I get around to opening a PR.