COVESA/vehicle_signal_specification #591: DIscrepancies in Pull Request guidelines

ppb2020, 2023-05-03:
The Pull Request guidelines here:

`https://github.com/COVESA/vehicle_signal_specification/blob/master/CONTRIBUTING.md`

mention using "Signed-off-by: Firstname Lastname firstname.lastname@example.com" for sign-off, while the Guidelines for submitters here:

`https://www.covesa.global/contribute`

mention using "Signed-off-by: Firstname Lastname <[firstname.lastname@example.com]>" instead.

Note the use of <> in the latter. IIUC, the use of <> is required.

As well, note that using an email address of "userid@example.com" seems to work fine. Is the guideline to use firstname.lastname a requirement or will any email address form be oOK

erikbosch, 2023-05-04:
Discussed shortly with Steve Crumb. "firstname.lastname@example.com" shall be seen just as an example, it could be "mr_big@hotmail.com" if you like to. The important part is that your real name is present before the e-mail address.

erikbosch, 2023-05-04:
The file at https://www.covesa.global/contribute is the master. There is work in progress to update it. The other is quick to update when the master is updated.

ppb2020, 2023-07-13:
Any update on this?

erikbosch, 2023-07-14:
I have updated our local [CONTRIBUTING.md](https://github.com/COVESA/vehicle_signal_specification/blob/master/CONTRIBUTING.md) so that it says:

_Start_

Every contribution must carry the following sign-off line with your real name and email address:

`Signed-off-by: Firstname Lastname <you@example.com>`

By supplying this sign-off line, you indicate your acceptance of the COVESA Certificate of Origin.

_End_

I have not heard anything more on update status for the [global COVESA guideline](https://www.covesa.global/contribute).
FYI: @paulboyes  - could you possibly check if there is anything in progress

paulboyes, 2023-07-15:
I will check for sure.  I do not see an issue with this.  The way it is worded now assumes everyone follows the same email naming convention.  Won't work.

paulboyes, 2023-07-19:
https://www.covesa.global/contribute has been updated.

erikbosch, 2023-09-11:
CLosing this one

closed 2023-09-11 by erikbosch