COVESA/vehicle_signal_specification #607: Integration of martime data

fulup-bzh, 2023-05-15:
In the process of moving to VSS, I need to integrate both BMS data and maritime data on top of VSS. While I did not find any signigicant open-source project for BMS data, for navigation there is an active project that is somehow similar to VSS  http://signalk.org/specification/1.2.0/doc/signalk.pdf

My goal is to merge both VSS&Signal-K specifications within the same dataset. This in order to provide a single dataset/API for BMS/J1939/Maritime datas.

Any idea would be more than welcome.

SebastianSchildt, 2023-05-17:
Just browsing  the PDF, a thought crossed my mind:  Do the "datapoints" described in Appendix A/B also exists in "machine readable" format somewhere? That would be maybe a starting point to "convert" them to VSS (as entry point into a joint stack). If it doesn't exist, if you were to "manually" "converting" them to vspec/writing them in vspec form on the other hand might be interesting for the signalk community because finally they WOULD have a machine readable version of their Datapoints

fulup-bzh, 2023-05-17:
You may find the Signal-k spec on github https://github.com/SignalK/specification/tree/master/schemas the main entry point is vessel.json then it point to sub-group ex for example groups/navigation.json Even if they use heavyly rely on json schema, signal tree model is not than different from VSS.

For low level signal I use canboat project https://github.com/canboat/canboat/blob/master/analyzer/pgns.xml (json is equivalent with less signal description)

erikbosch, 2023-06-20:
Meeting notes:

- No action point, has been discussed, closing it

erikbosch, 2023-06-20:
Please check https://wiki.covesa.global/display/WIK4/Defining+the+COVESA+data+modeling+strategy+and+its+associated+artifacts as it tries cover also other domains. Add comments/question if wiki

closed 2023-06-20 by erikbosch