COVESA/vehicle_signal_specification #626: Depressed PedalPosition

mikehaller, 2023-06-27:
`Vehicle.Chassis.Chassis.Accelerator.PedalPosition` is currently described as:

```
Accelerator.PedalPosition:
  datatype: uint8
  type: sensor
  min: 0
  max: 100
  unit: percent
  description: Accelerator pedal position as percent. 0 = Not depressed. 100 = Fully depressed.
```

Using the term `depressed` is probably a mistranslation: `gedrückt` (pushed) or `bedrückt` (depressed) in German

The description should read:

```
Accelerator pedal position as percent. 0 = Released. 100 = Fully pushed down.
```

References:
<https://github.com/COVESA/vehicle_signal_specification/blob/9481aef876f117f3cdec2eba867fbd43f27060b6/spec/Chassis/Chassis.vspec#L183>

petervolvowinz, 2023-06-27:
According to Merriam-Webster:
depress:
to press down
to cause to sink to a lower position.

erikbosch, 2023-06-27:
Meeting notes: Please comment on preferred wording. if possible refer to terms in a standard

Some notes: In J1939 they talk about closed pedal (when pedal is not pressed, vehicle idling) and open pedal.

In (British) English "depressed pedal" exists now and then:

https://www.volkswagen.co.uk/en/owners-and-services/my-car/warning-light/electric-depress-the-brake-pedal.html

_A red depress the brake pedal light means you should stop the car as soon as it's safe_

https://www.kia.com/content/dam/kia2/in/en/content/seltos-manual/topics/chapter6_3_2.html

_Manual Transmission - Depress the clutch pedal fully and shift the transmission into Neutral._

https://www.bosch-presse.de/pressportal/de/en/bosch-esitronic-diagnostic-software-offers-a-helping-hand-when-working-on-electric-and-hybrid-vehicles-255488.html

_For example, on the Model S and X, the brake pedal must be depressed to put the vehicle into the mode that allows access to the CAN diagnostic functions._

mikehaller, 2023-06-29:
Wow, i'm kind of ashamed for not doing proper research on this. Thanks for clarification!

I'm still a bit confused about the J1939 example though:
*closed pedal* is the same as *pedal not pressed*?
So `closed == released == not depressed` ?

closed 2023-07-11 by erikbosch