COVESA/vehicle_signal_specification #635: Deprecation and Removal of OBD branch?

erikbosch, 2023-07-31:
A topic that comes up now and then is removal of the [OBD](https://github.com/COVESA/vehicle_signal_specification/blob/master/spec/OBD/OBD.vspec) branch. Sometimes the rationale is that COVESA should not care about diagnostics data at all, sometimes that there is no reason to duplicate signals, if we need signals concerning for examples oxygen sensors and trouble codes they should exist in the "normal" tree.

This issue intends to collect feedback on the topic as input for decision. Is there someone that is strongly against removing the OBD branch? If so, why? Could it be mitigated by for example adding new signals in other trees?

To start the discussion - would this be a feasible way forward?

- In VSS 5.0 we mark the OBD branch as deprecated
- In VSS 6.0 we remove it.
- Until VSS 5.0 (or latest at VSS 6.0) everyone that wants/needs information that currently exist only in OBD-branch must create corresponding signals in some other branch.

ppb2020, 2023-07-31:
Seems like a good plan to me.

nickrbb, 2023-08-01:
I agree, removing duplication of signals is a good idea. I would suggest we make efforts to port all of the signals though, as VSS has a much wider consumer base than it does active contributor base. It would also make VSS a much better alternative to OBD (since it would host a superset of OBD signals).

Furthermore, it might also be a worthwhile effort though to provide a mapping in some form or another (possibly just a readmeOBD.md file) of OBD signals to their VSS equivalents, at least in v6, to help e.g. OBD experts who are also VSS newbies. This could prevent people bringing in contributions later on introducing a specific OBD branch and signal that already exists in VSS but which is not so easy to find/map.

SebastianSchildt, 2023-08-02:
It might make sense, to still provide OBD as an "offical" overlay, as it is still in widespread use, also for simple telemetry applications.

Other than that I'd think that maybe DTC codes should be somehow migrated, as those are really the basis for many "first-shot" diagnostics, but those might be split into the different branches.

adobekan, 2023-08-08:
> It might make sense, to still provide OBD as an "offical" overlay, as it is still in widespread use, also for simple telemetry applications.
>
> Other than that I'd think that maybe DTC codes should be [...]

+1 We should definitely move them to an overlay.

SebastianSchildt, 2023-08-15:
Meeting 8/23

 - "how to deprecate" correctly, so that people can upgrade VSS if they want while still being able to rely on OBD signals if they need them
 - seems "doable" here like Erik lined out above (it is harder when changing metadata such as dataypes/units, but this i not the case here)