COVESA/vehicle_signal_specification #650: possible bug in units.yaml: nm/km

brielb, 2023-09-14:
https://github.com/COVESA/vehicle_signal_specification/blob/7d8848f7ab39a8fc335f25a85e097cfd41abcdc8/spec/units.yaml#L207C2-L210C18

there is a unit called nm/km, which is resolved to nanometers per km, whoch actially resolves to a constant: 10^(-12).

I assume it's reverring to kilometers and nautical miles, where the latter wouldn't make sense for a land vehicle, should rather be imperial miles.

erikbosch, 2023-09-14:
The unit `nm/km`originally comes from https://github.com/COVESA/vss-tools/pull/174 (the file was moved to this repo some time ago). From the history of that PR it seems that the value refer to physical wear, like how much wear on the tires or your break discs are you currently experiencing, but then domain should possibly better be "wear per distance" or similar that gives some background. So for that purpose it could possibly make sense. We do however not have signals in standard catalog actually using it. 

FYI: @adobekan @tobiask84 - you were involved in the original PR, any comments.

erikbosch, 2023-11-29:
Closing this, not considered a bug

closed 2023-11-29 by erikbosch