COVESA/vehicle_signal_specification #661: Revised Description of "PowerOptimizeLevel" Parameter with Suggested Addition

UmangSharmaMobis, 2023-09-26:
PowerOptimizeLevel:
  datatype: uint8
  type: actuator
  min: 0
  max: 10
  description: |
    Power optimization level for this branch/subsystem.
    A higher number indicates more aggressive power optimization.
    Level 0 indicates that all functionality is enabled, no power optimization enabled.
    Level 10 indicates most aggressive power optimization mode, only essential functionality enabled.

    Note: This definition does not include an automatic optimization level. Some specific specifications or features may be handled automatically by the OEM during runtime.

    Suggested Addition:
    Level 11 - Auto: This level enables automatic optimization, allowing the system to dynamically adjust the power optimization level based on specific runtime conditions or features handled by the OEM.
```

I've included your suggested addition for Level 11 - Auto, which enables automatic optimization, allowing the system to dynamically adjust the power optimization level based on specific runtime conditions or features handled by the OEM.

erikbosch, 2023-09-26:
Meeting notes:

- Umang introduced the issue
- Please discuss, to be discussed/decided next meeting

SebastianSchildt, 2023-09-26:
I am not sure, if we should add "magic numbers". That feels a lot like lower level CAN stuff, which we like to "transcend" with VSS.

Maybe another approach would be to have a bool *AutoPowerOptimze*. If it is `true` convetion might be that *PowerOptimizeLevel* just reflects the current status chosen by the Auto Optimizer (or maybe stays at 0, maybe an implemtation detail), and if somebody chooses a manual level by writing to *PowerOptimizeLevel* an implementation might set *AutoPowerOptimze* to `false`

But that is just one example, my main point is, I  think VSS should not have "magic numbers" wherein "0. to 10" means "0 to 10", but "11" or "-5" in the same signal mean some special condition.

UmangSharmaMobis, 2023-10-03:
I agree with you as well, Sebastian. My initial thought was to introduce an additional level as '11' and designate it as an automatic level. However, your suggestion of using 'AutoPowerOptimize' with a boolean datatype is a better approach and will effectively address the issue.

erikbosch, 2023-10-03:
Is the intention then to just have a single `AutoPowerOptimize`, or one in every place we today have `PowerOptimizeLevel`?

erikbosch, 2023-10-03:
Meeting notes:

- Umang to create pull request
- Then please review

UmangSharmaMobis, 2024-01-24:
AutoPowerOptimize signal has been added in #670 issue can be closed

closed 2024-01-24 by UmangSharmaMobis