COVESA/vehicle_signal_specification #662: Does VSS-syntax only support a single root?

erikbosch, 2023-09-27:
Today we mention here and there in the documentation that the VSS catalog forms a tree, and vss-tools actually gives an error if you have more than one root. But we do not really say anywhere that a model using VSS-syntax must have a single root, i.e. that something like below is an invalid VSS model. 

If we want to keep the idea that there can only be a single tree and a single root I think we should improve documentation so it clearly states that. If we do not think that is a reasonable limitation we should start looking at how vss-tools can support multiple trees.

(Whether a VSS model must have exactly one root or not does not necessarily affect downstream implementations, as an implementation may support access to multiple models in parallel)

```
a:
  type: branch
  description: Branch A.

a.b:
  datatype: float
  type: sensor
  unit: km
  description: A sensor.
  
c:
  type: branch
  description: Branch A.

c.d:
  datatype: float
  type: sensor
  unit: km
  description: A sensor.
```

ppb2020, 2023-09-27:
FWIW, in our implementation, we do rely on the fact that there is a single root. It probably would not be a big deal to adapt,but that would require a few changes, some of which may not be trivial.

erikbosch, 2023-10-03:
Similar to the discussion on node/signal names (mandatory rules vs. recommendations) we could actually think of a dual approach:

- VSS tools (and VSS syntax definition) shall support multiple roots
- But it is recommended to only have 1 root, and that would be a requirement for the standard cataalog (i.e. with `--strict` you would get an error if having multiple roots)

commit 93550c39 (2024-02-20): Update documentation

Clarifies string/float/double.
Reformat recommendatations to state that they also are rules for VSS catalog
Remove enum recommendation

Fixes #705
Fixes #706
Partially solves #700
Fixes #662
commit db3b08bb (2024-05-08): Update documentation

Clarifies string/float/double.
Reformat recommendatations to state that they also are rules for VSS catalog
Remove enum recommendation

Fixes #705
Fixes #706
Partially solves #700
Fixes #662
closed 2024-02-20 by erikbosch