COVESA/vehicle_signal_specification #678: Wheel angular velocity

SebastianSchildt, 2023-10-26:
### Discussed in https://github.com/COVESA/vehicle_signal_specification/discussions/677

<div type='discussions-op-text'>

<sup>Originally posted by **PKersting** October 26, 2023</sup>
Wouldn't it be more suitable to use angular velocity (rad/s) instead of rotational velocity in km/h for the vehicle speed? 
see https://github.com/COVESA/vehicle_signal_specification/blob/master/spec/Chassis/Wheel.vspec</div>

SebastianSchildt, 2023-10-26:
Thanks for posting this @PKersting! I converted this to an issue, as I think so far nobody used discussions (We are using issues for that). 

Here is the discussion from when the datapoint was created https://github.com/COVESA/vehicle_signal_specification/pull/464

It seems at the time I had similar doubts as you, but at the time available information pointed to a "linear speed" being more useful. Are there any (public) examples in which context or standards a rad/s value is used/better

Especially in this case it is not a "trivial" conversion for an application/service using the signal  like km/h vs m/s as at the dimensions of the tire come into play

erikbosch, 2023-10-27:
It is also partially related to the discussion in #675 on derived data, would we benefit from having multiple signals for different use-cases? I.e is there a value in having both? If you know tire dimension you can convert between them (even if I do not know if you need to add a small correction factor to consider dynamics of the tire, tire pressure, temperature and so on...)

For reference, in Andoid VHAL they instead have tick-counter and "micrometers per tick", which could be used to derive wheel speed in km/h, but not rad/s unless you know "number of ticks per rotation" 

https://android.googlesource.com/platform/hardware/interfaces/+/refs/heads/main/automotive/vehicle/2.0/types.hal#488 

As pointed out by Sebastian - it would be good to know more about actual use cases. My understanding is that some internal functions (like ABS) typically work on change in ticks rather than km/h or rad/s. On the other hand, if a vehicle use wheel speed as input for `Vehicle.Speed` there must somewhere be a function that converts the raw value (change in ticks) to km/h (or any other speed unit)

PKersting, 2023-10-27:
I think you can argue one way or the other, depending on the use case. If km/h is to remain, it would be good in my point of view to change the description from "Rotational speed of a vehicle's wheel." to "Speed of a vehicle's wheel on ground level." "Rotational" does not seem to be the right term in this context.

erikbosch, 2024-02-20:
Discussed 2024-02-13, see https://wiki.covesa.global/pages/editpage.action?pageId=48529442

No problem to add another signal

PR #726 (2024-03-11): Add wheel angular speed
Fixes #678
Related to #677 

I suggest that we add a new signal rather than "refactoring" the old, as there might be use-cases where linear speed is useful, like if you have a passive TPMS system (and different wheel sizes) or if you use the average speed of individual wheels) with different size)  to calculate `Vehicle.Speed`

Also changing description of existing signal, but semantics should be the same.
I used `degrees/s` as unit as that is the only unit we currently have for angular velocity.
As I see it it would not be a problem for us to add `rad`and `rad/s` to th [... cut]
closed 2024-03-11 by erikbosch