COVESA/vehicle_signal_specification #683: Types and datatypes

doganulus, 2023-11-03:
I would like to share my observations about VSS types and datatypes. The current classification is as follows:

```yaml
type: [sensor, actuator, attribute, property, struct, branch]
datatype: [boolean, string, numbers, arrays]
```

I have two questions here:
- First, the difference between types `attribute` and `property` is unclear to me. Let me try to summarize my understanding below. Is that right?
  - `sensor` denotes a dynamic (temporal) signal not directly controllable by the system -- fast change
  - `actuator` denotes a dynamic (temporal) signal directly controllable by the system -- fast change
  - `attribute` denotes a dynamic? configuration variable -- thus, runtime-changeable but not often.
  - `property` denotes a static? configuration parameter? -- runtime-constant

- Second, I understand that `struct` is a relatively new keyword for the VSS project. My initial assessment would place the structural elements of `struct` and `branch` under `datatype` as I could not find a functional description for `struct` as above. The same is true for `branch` where the difference is that structs are packed, and branches are not. Therefore, what would say about the following classification that moves them under datatypes? 

```yaml
type: [sensor, actuator, attribute, property]
datatype: [boolean, string, numbers, arrays, struct, branch]
```

Even more, taking full liberty to ignore all necessary backward compatibility issues (probably the project cannot do very easily), I would prefer the key `type` instead of `datatype` for compatibility with other specification formats, mainly JSON Schema, and prefer another term, like `kind` or `class`, to denote its usage in the system.

```yaml
kind: [sensor, actuator, attribute, property]
type: [null, boolean, string, numbers, tuple, list, struct, branch]
```

And, a straightforward mapping to JSON types may be valuable as I would map `tuple` and `list` to JSON arrays as well as `struct` and `branch` to JSON objects. But I admit I don't have much prior experience with other formats that VSS can be converted.

jsyvanen, 2023-11-05:
>I understand that struct is a relatively new keyword for the VSS project. My initial assessment would place the structural elements of struct and branch under datatype as I could not find a functional description for struct

I was also perplexed by this since the current [docs](https://covesa.github.io/vehicle_signal_specification/rule_set/data_entry/data_types_struct/) for `struct` seem to use `type` and `datatype` interchangeably in a few places. But I believe the intention is that `struct` is a `datatype` and not a `type` in the sense of `sensor`, `actuator`, `attribute`.

>the difference between types attribute and property is unclear to me

Relating to the above definition of `struct` as a `datatype`, my best interpretation at the moment is that `property` is a "type" intended to be used only inside `struct` definitions. Indeed it is the only place where the `property` "type" is referenced within the repo at the moment.

It would be great to have the understanding confirmed/disconfirmed. If the above is correct I would like to understand the intended usage of the `property` "type" in the context of `struct` datatypes.

jsyvanen, 2023-11-21:
@erikbosch thanks for #688 . Would you be able to comment on the specific question above about intended use of `property` ?

erikbosch, 2023-11-21:
@jsyvanen - you can check som struct examples in vss-tools for testing, like the one in https://github.com/COVESA/vss-tools/blob/master/tests/vspec/test_structs/TestBranch1.vspec

As of today ` type: property` is intended to be used only to specify members of a struct type. So the snippet below

```
NestedStruct:
  type: struct
  description: "A struct that is going to be used within another struct - Nested"

NestedStruct.x:
  type: property
  description: "x property"
  datatype: double
  min: -10

NestedStruct.y:
  type: property
  description: "y property"
  datatype: double
  max: 10

NestedStruct.z:
  type: property
  description: "z property"
  datatype: double
  default: 1
```

Would if represented in for example C++ be something like

```
struct NestedStruct {
  double x;
  double y;
  double z;
};
```

I.e. the VSS properties correspond to C++ struct members.

PR #688 (2023-11-21): Align usage of the word type.
This PR intends align terms used in documentation

* If we refer to a VSS data type, the term "datatype" is used
* If we refer to the VSS node attribute "type" that specifies if a node is a branch, sensor, actuator, struct, ... we use the term "type"

Fixes #683
commit f1bb8633 (2023-11-21): Align usage of the word type.

This PR intends align terms used in documentation

* If we refer to a VSS data type, the term "datatype" is used
* If we refer to the VSS node attribute "type" that specifies if a node
  is a branch, sensor, actuator, struct, ... we use the term "type"

Fixes #683
closed 2023-11-21 by erikbosch