COVESA/vehicle_signal_specification #689: Need for a 5.0 release this year?

erikbosch, 2023-11-14:
We have decided to release 4.1 before year end. It will contain all minor (not backward incompatible changes) from  master, see current content at 4.X branch. In addition to those changes we have 5-10 changes changing existing signals. As they are very few we are considering NOT to release 5.0 this year, but possibly wait until April/May next year (close to the AMM).

But if there are someone that would see a big value in a 5.0 already this year we could possibly create a release. If you want a 5.0 already this year please add a comment to this issue.

@ppb2020 - you have added some incompatible changes to VSS-tree. Do you or your organization need a 5.0 (including the changes) already this year, or are you ok with waiting until spring?

ppb2020, 2023-11-14:
We don't need the changes right now, but we'd also prefer not to wait till next spring. However, we have a way forward in the short term if need be: we can take a selection of signals of interest from v5.0 and apply them over v4.1 using overlays, as we had done previously to bring in upcoming v3.0 (as of yet unreleased at the time) signals we needed on top of v2.1, and create a BlackBerry Signal Catalog. By doing this, we kind of future-proof our eventual move to a later release.

Thus, we'll adapt to whatever release COVESA deems best for their needs.

erikbosch, 2023-11-28:
Discussed in recent VSS meeting - conclusion let us put VSS 5.0 for hold right now, i.e. only release 4.1. But be prepared to release 5.0 when considered needed.

erikbosch, 2024-02-09:
No requests for 5.0 yet, but we should better decide.

Suggestion:

- Prepare for a 5.0 release around AMM (April)
- Plan for the last 4.X release 6 months later (i.e. around fall AMM, October)
- But do we need another 4.X release in between?

erikbosch, 2024-02-13:
MoM:

- Sebastian: We should release 5.0, when we release a 5.X we should check if we also release a 4.X
- Nick: Release around AMM would be good

ppb2020, 2024-03-12:
This was opened in November last year and there weren't enough changes to justify a release at that time. What's the "count" as of today? Are there enough changes to justify a 5.0 release? Same question for 4.x.

Indeed, if there are enough changes, then releasing just before the AMM (so that the release can be announced there) would be good timing.

closed 2024-09-19 by erikbosch