COVESA/vehicle_signal_specification #694: In-Vehicle Notification signal for Infotainment

UmangSharmaMobis, 2023-11-28:
In-Vehicle Notification signal for Infotainment

Notification Signal branch for all notification actions
 
Notification on/off
Notification type

Notification priority, data definition to display information and push notification related 

Signal definition for:

- Notification for sensors detecting events like low tire pressure, engine overheating, or low fuel levels.
- Notification for vehicle's internal systems in response to detected issues or significant events, providing details
- is Push  Notification

erikbosch, 2023-11-30:
Sorry that we did not have time to discuss this issue at the last meeting. We should discuss it next week.

Is it the tell-tale (symbols) you are interested in, or the underlying values? We have today some boolean values that typically could be the input for a tell-tale, like `Vehicle.Powertrain.FuelSystem.IsFuelLevelLow`, but there is likely not a 1:1 connection to the actual tell-tale. As an example, all tell-tales may be lit for a short period when you start the engine to show that they are functional, and maybe a tell-tale like `isFuelLevelLow` will not be lit unless ignition is on.  

Or rather a more generic one, like Diagnostic Trouble Codes? As part of deprecating/removing the DTC branch we have talked about that it would make sense to represent DTCs somewhere in the tree. Could possibly be a case where structs could make sense to describe the content of each item.

UmangSharmaMobis, 2024-01-24:
Apologies for the delay in my response due to an extended holiday in India. I have consulted with my team, and they are currently in the process of reviewing. I will reopen the matter and furnish additional details shortly.

closed 2024-01-24 by UmangSharmaMobis