COVESA/vehicle_signal_specification #724: Vehicle Error Codes

JonathanForce, 2024-02-19:
Hello,

For our recreational vehicles, we would like to be able to retrieve the various error codes that the vehicle may encounter.

I've noticed that some signals have an isError boolean, for example for ABS (ABS.IsError) or ESC (ESC.IsError), but this isn't precise enough for our purposes.

In TractionBattery there's an `ErrorCodes` signal, but it's specific to the battery.

Would it be possible to have the signal below directly in Vehicle.vspec?
```
ErrorCodes:
  datatype: string[]
  type: sensor
  description: Current error codes related to the vehicle
```

This would contain all the error codes encountered by the vehicle and we could inform the user of the problems and actions to be taken.

What do you think?
Is there another/better option?

erikbosch, 2024-02-20:
We have had Diagnostic Trouble Code (DTCs) as part of the OBD branch for a while, and as part of the OBD deprecation they were recently migrated/copied to a branch of it's own at https://github.com/COVESA/vehicle_signal_specification/blob/master/spec/Vehicle/Diagnostics.vspec

```

Diagnostics.DTCList:
  datatype: string[]
  type: sensor
  description: List of currently active DTCs formatted according OBD II (SAE-J2012DA_201812) standard ([P|C|B|U]XXXXX )
```

JonathanForce, 2024-02-21:
Thank you @erikbosch, we will use this signal for our errors.
I close the issue.

closed 2024-02-21 by JonathanForce