COVESA/vehicle_signal_specification #733: Inconsistent terminology - attributes, members, elements, meta data

ppb2020, 2024-03-13:
I have noticed that the documentation and tools are not consistent in using terminology to describe the elements that define the meta data for a node (such as type, datatype, description, etc.).

Under the document for Branches, these are referred to as "elements". Under the document for Data Entry, the more general "metadata" term is used. Under the document for "Sensors & Actuators", the term "members" is used. Lastly, vss-tools refer to the same as "attributes", a usage that can be confused with nodes that define attribute types.

I propose rationalizing this by using the term "element" everywhere and, perhaps, metadata in some places where we refer in documentation to the set of elements used to define a node.

I can start putting together a set of changes to align this terminology across the specification, its documentation, and the tools. I'd probably fix up a few typos I have come across as well at the same time... Any objections?

erikbosch, 2024-03-19:
No objections, rather the opposite. We need to consider what changes that will be needed (long term) in vss-tools, as some arguments there include the term "attribute". I suggest us to discuss and try to agree on terms in VSS-meeting before you start doing search/replace everywhere.

erikbosch, 2024-03-19:
MoM:

- Please discuss; do you think `element` is a good name
- Daniel: There is room for improvement, could be considered as `label`, or `keys`
- Continue discussion next week

ppb2020, 2024-03-26:
Perhaps a topic for the AMM?