COVESA/vehicle_signal_specification #734: typo in vspec v3.0 makes vss-tools 4.x fail

lukasmittag, 2024-03-18:
I found something really interesting in vspec 3.0 there is a typo 
``` Actuator ```  instead of ```actuator```  
https://github.com/COVESA/vehicle_signal_specification/blob/525e2bd00ddf061851bdc75e849178e5d3ad5833/spec/Powertrain/Battery.vspec#L229C9-L229C17 
and this makes load_tree impossible for vspec definiton 3.0 
Suggested fix would be to allow upper and lower case here 
https://github.com/COVESA/vss-tools/blob/839b0d087ecb6419372476fb99a7d8c160c7d13b/vspec/model/constants.py#L284C29-L284C30

erikbosch, 2024-03-18:
We had a discussion around a year ago to clarify that VSS names are case sensitive, see https://github.com/COVESA/vehicle_signal_specification/pull/569 and as part of that we changed documentation and vss-tools. 

So far we have not had any ambition to keep support for older VSS versions in VSS-tools, the assumption has sort of been that you use the VSS-tools version related to the VSS-version you are using. This issue is a good starting point to discuss if we better should change this approach; some alternatives below

- Should we continue just supporting "latest and greatest VSS version and standard catalog"
- Should we automatically supporter older supported VSS_version (v3.0 may anyway be too old to be considered supported)
- Should we explicitly add support for selected older versions; like `--version 3.1` to accept 3.1-syntax rather than "latest syntax"

ppb2020, 2024-03-26:
I would not have strong objections to requiring alignment between catalog version and tool version. The main advantages of this would be to reduce testing and simplicity of use.

Otherwise, we could support some form of backwards compatibility, where the latest version of the tools would support older versions of the catalog. This seems onerous and may not be possible in the long term.

With a backwards compatible approach or an approach where one specifies the syntax version, could this instead be done by looking at the existing version entries in the catalog (i.e., Vehicle.VersionVSS)?