COVESA/vehicle_signal_specification #754: Insurance relevant Parameters

StefanSellschopp, 2024-07-09:
[09072024_Covesa_Allianz_UseCasesADAS_Risk_LiabilityData_Table_V8.xlsx](https://github.com/user-attachments/files/16145449/09072024_Covesa_Allianz_UseCasesADAS_Risk_LiabilityData_Table_V8.xlsx)

the first tab describes relevant data points (AggregationDataPoints),  (the tabs on the right describes specific situations and can be ignored)
the first ones include the VSS parameters I could find in VSS4.2 - any additions?
the lower ones are missing from my perspectives - any thoughts?

SebastianSchildt, 2024-07-16:
The attached xls has more, but for convencience and a quick glance: 
[image]

[image]

[image]

SebastianSchildt, 2024-07-16:
So there is acceleration in Vehicle:

```yaml
Acceleration:
  type: branch
  description: Spatial acceleration. Axis definitions according to ISO 8855.

Acceleration.Longitudinal:
  datatype: float
  type: sensor
  unit: m/s^2
  description: Vehicle acceleration in X (longitudinal acceleration).

Acceleration.Lateral:
  datatype: float
  type: sensor
  unit: m/s^2
  description: Vehicle acceleration in Y (lateral acceleration).

Acceleration.Vertical:
  datatype: float
  type: sensor
  unit: m/s^2
  description: Vehicle acceleration in Z (vertical acceleration).
```

Same for gyroscope:

```
#
# Spatial Rotation
#
AngularVelocity:
  type: branch
  description: Spatial rotation. Axis definitions according to ISO 8855.

AngularVelocity.Roll:
  datatype: float
  type: sensor
  unit: degrees/s
  description: Vehicle rotation rate along X (longitudinal).

AngularVelocity.Pitch:
  datatype: float
  type: sensor
  unit: degrees/s
  description: Vehicle rotation rate along Y (lateral).

AngularVelocity.Yaw:
  datatype: float
  type: sensor
  unit: degrees/s
  description: Vehicle rotation rate along Z (vertical).
```
Basically insurance side there would needed to be the added "non-functional  requirement" for a system to provide this in the desired frequency and range

*Dashcam data*: In VSS we would not "model" video data (as a datatype/stream), but certeinly the existence of a dashcam and a "location" to the ringbuffer may be modelled, see i.e. https://covesa.github.io/vehicle_signal_specification/rule_set/data_entry/data_types/#data-streams

*Pedestrian Protection*:Does this mean some form of pedestrian protection has been deployed, like an airbag? (or this kind of "reactive armor" kind of hoods?), or does this mean existence of some system?

*descent/ascent*: What is meant here? Whether the vehicle is on a slope?

SebastianSchildt, 2024-07-16:
And Outside AirTemperature is

Vehicle.Exterior

```yaml
AirTemperature:
  datatype: float
  type: sensor
  unit: celsius
  description: Air temperature outside the vehicle.
```

tguild, 2024-07-16:
@StefanSellschopp for easier iterating the spreadsheet, I converted to a Google Sheet as discussed. I've made you, @SebastianSchildt and myself initial editors, expand as appropriate.

https://docs.google.com/spreadsheets/d/1gnaP8dUER4uOlgC9htJssW9rRdg0D_FA/edit?gid=1736927769#gid=1736927769

StefanSellschopp, 2024-07-16:
Internal

OK

https://docs.google.com/spreadsheets/d/1gnaP8dUER4uOlgC9htJssW9rRdg0D_FA/edit?gid=1736927769#gid=1736927769

StefanSellschopp, 2024-07-16:
pedestrian protection means the sensor and upward movement of the hood, in case of a pedestrian accident.
ascent/decent: is a vehicle moving upwards or downwards

erikbosch, 2024-07-16:
MoM: 

- Prioritize this topic for next meeting.

StefanSellschopp, 2024-07-29:
I would suggest to add crash related parameters in one PR
vehicle.body.rollover
datatype: boolean
type: sensor (?)
description: vehicle detects roll-over event - TRUE trigger

vehicle.body.pedestrianDetection
datatype: boolean
type: sensor (?) 
description: vehicle detects an impact with a pedestrian, the hood is pushed upwards  - TRUE trigger

erikbosch, 2024-08-07:
@StefanSellschopp - as discussed in the meeting I added a small tutorial for creating a Pull Request at https://github.com/COVESA/vehicle_signal_specification/wiki/Creating-a-VSS-Pull-Request.

I used your signals above as example, visible in (my fork internal) PR https://github.com/erikbosch/vehicle_signal_specification/pull/16. Feel free to experiment with creating a PR yourself.

When the PR is created I see two things that may need to be discussed

- Where to put the new signals. Not obvious as I see it.
- Define detailed semantics, like when shall they be true, when shall they be reset to false.

StefanSellschopp, 2024-08-07:
Internal

Thanks, I will give it a try…

I used your signals above as example, visible in (my fork internal) PR erikbosch#16<https://github.com/erikbosch/vehicle_signal_specification/pull/16>. Feel free to experiment with creating a PR yourself.

StefanSellschopp, 2024-08-16:
@erikbosch 
i did try to do a fork. Then tried to copy your changes. my file ending does not say .vspec - how do I create this?
also I was not able to create a "folder" to include another vspec file on CrashNearCrash. I manually copied the header lines and then added some parameters. I think we should move the CrashManagement parameters here as well.
Also added slope in the "chassis" file
not sure if this correct.

StefanSellschopp, 2024-08-27:
@SebastianSchildt : any ideas on how to proceed on this? what do i need to change?
eg. https://github.com/StefanSellschopp/vehicle_signal_specification/blob/master/spec/Crash_NearCrash

StefanSellschopp, 2024-09-05:
we can discuss in our regular calls, no problem. Perhaps you have a suggestion?