COVESA/vehicle_signal_specification #795: Vss signal not able to map in json

sanyakamra1996, 2024-12-12:
Hi,

I have a signal in VSS:

```
Vehicle.Chassis.Axle.Row1.Wheel.Left.Tire.Pressure:
  datatype: unit16
  type: sensor
  description: Front left tire pressure.
  dbc2vss:
    on_change: true
    signal: TPM_FLR
```
However, when I build VSS, this signal doesn’t show up in the JSON.

Other signals like Vehicle.Body.Trunk.Rear.IsOpen or Vehicle.Cabin.Door.Row1.PassengerSide.IsOpen work fine and display the DBC signal in the JSON.

For the above signal, I even tried using Vehicle.Chassis.Axle.Row1.Wheel.Left.Tire.IsPressureLow, but it still doesn’t appear in the JSON.

The existing JSON seems to include default entries, like this:
```

"Pressure": {
  "datatype": "uint16",
  "description": "Tire pressure in kilo-Pascal.",
  "type": "sensor",
  "unit": "kPa"
}
```
Could you help me understand why the signal isn’t being reflected properly in the JSON?

erikbosch, 2024-12-13:
Hi!

How did you generate the JSON? I tried:

* Adding an overlay file `overlay.vspec`

```
Vehicle.Chassis.Axle.Row1.Wheel.Left.Tire.Pressure:
  datatype: uint16
  type: sensor
  description: Front left tire pressure.
  dbc2vss:
    on_change: true
    signal: TPM_FLR
```

Then generating JSON using installed vss-tools, the overlay and the extra attribute `dbc2vss`:

`erik@debian6:~/vehicle_signal_specification$ vspec export json --vspec spec/VehicleSignalSpecification.vspec --overlays overlay.vspec --pretty -o hej.json --extended-attributes dbc2vss`

Then looking in the JSON the overlay seem to have been correctly applied.

```
                         "Tire": {
                            "children": {
                              "IsPressureLow": {
                                "datatype": "boolean",
                                "description": "Tire Pressure Status. True = Low tire pressure. False = Good tire pressure.",
                                "type": "sensor"
                              },
                              "Pressure": {
                                "datatype": "uint16",
                                "dbc2vss": {
                                  "on_change": true,
                                  "signal": "TPM_FLR"
                                },
                                "description": "Front left tire pressure.",
                                "type": "sensor",
                                "unit": "kPa"
                              },
                              "Temperature": {
                                "datatype": "float",
                                "description": "Tire temperature in Celsius.",
                                "type": "sensor",
                                "unit": "celsius"
                              }
                            },
                            "description": "Tire signals for wheel.",
                            "type": "branch"
                          }
```

sanyakamra1996, 2024-12-17:
Hi @erikbosch 
what is the version of vss-tool you used?
I used 4.1
```
python3 vss-tools-4.1/vspec2json.py \
    -e vss2dbc,dbc2vss,dbc \
    -o /vss/bz4x_overlay.vspec \
    -o /vss/occupant_branch.vspec \
    -o /vss/vehicle_trip.vspec \
    -o /vss/driver_branch.vspec \
    -o /vss/driver_signals.vspec \
    -o /vss/tpms.vspec \
    -u vehicle_signal_specification-4.1/spec/units.yaml \
    --json-pretty vehicle_signal_specification-4.1/spec/VehicleSignalSpecification.vspec \
    /vss/vss_bz4x_dbc.json \
    --extended-attributes dbc2vss
```
But wasn't able to get in json still

erikbosch, 2024-12-18:
I used a local build from the `master` branch, which is quite similar to the released `5.0`.

It seems that you are experiencing this bug: https://github.com/COVESA/vss-tools/pull/346

That was fixed in 4.2, so updating to 4.2 could be an option (In 5.0 the CLI was restructured and some checks became stricter, so would be a bigger hurdle to upgrade to 5.0). Off course you could also try to patch your 4.1-version with the changes from the PR.

A side note, in your example you use both `-e` and `--extended-attributes`. That is the same argument, so I think the second one will "overwrite" the first one.