COVESA/vehicle_signal_specification #802: Support enum types similar to structs to avoid redefinition of common values

doosuu, 2025-01-22:
In some cases it would make a lot of sense to support enum types similar to structs because multiple sensors report the same status e.g. `OPEN`, `OPENING`, `CLOSED`, `CLOSING`could apply for windows sensors and door sensors. Instead of re-defining the allowed values every time, one would be able to define them centrally.

At the same time, it would be possible to define a base datatype to use for the enum representation, together with key-value mappings to allow for optional on-the-wire optimization of the type.

e.g.
```yml

Types.OpenState:
  type: enum
  datatype: uint8
  description: Opening state of openable sensors.
  valueTable:
    OPEN: 0
    OPENING: 1
    CLOSED: 2
    CLOSING: 3

```

erikbosch, 2025-01-23:
I am in favor of it as it increases reusability. One thing to discuss if we decide to implement is how it shall affect output formats. Not all downstream projects support "types" (structs) yet, so if we would refactor the catalog to use this new style we need to discuss how various output formats (yaml, csv, json, franca, ...) shall be affected. Like do we need to have a compatibility mode in vss-tools that converts the new style to the old "enum" style for tools that only support the old style.

A partially related feature would be the possibility to use a similar concept to specialize/restrict numeric types by min/max. Taking latitude as an example where we as of today in VSS have descriptions like:

```
Navigation.DestinationSet.Latitude:
  datatype: double
  type: actuator
  min: -90
  max: 90
  unit: degrees
  description: Latitude of destination in WGS 84 geodetic coordinates.
```

That could potentially also benefit from a reusable type, for example if we have multiple signals using latitude.

```
Navigation.DestinationSet.Latitude:
  datatype: Types.LatitudeDegrees
  type: actuator
  description: Latitude of destination in WGS 84 geodetic coordinates.

Types.LatitudeDegrees
  type: datatype
  datatype: double
  min: -90
  max: 90
  unit: degrees
```

erikbosch, 2025-01-23:
@jdacoello - could you check if this fits your ideas on model representation

jdacoello, 2025-01-23:
> [@jdacoello](https://github.com/jdacoello) - could you check if this fits your ideas on model representation

Yes, I have mentioned this already while discussing VSS limitations. Having a pre defined list of options as enum makes sense in many cases. However, I am not in favour of continuing to add features to the current VSS approach. 
Main reason is because we have reached its limits when it comes to the expressiveness. 
Any addition, as you know, requires tooling and possibly refactoring.
I would rather put a collective effort to replace the language with something that support this feature (and a few others) out of the box.

In `GraphQL` this is easily done like this:

```graphql
type Door {
  windows: [Window]
  openState: OpenStateEnum
}

type Window {
  openState: OpenStateEnum
}

enum OpenStateEnum {
  OPEN
  OPENING
  CLOSED
  CLOSING
}
```

As you can see, it is clean and easy to follow. In that example:
* Type and enum names must be unique
* Field names inside a particular type must be unique.
* Values inside a particular enum must be unique.
* A property might be accesible with multiple paths. For example:
```graphql
query {
  window {
    openState
  }
}
```

```graphql
query {
  door {
    windows {
      openState
    }
  }
}
```

jdacoello, 2025-01-24:
If we really want to add this feature in the current VSS approach, I would suggest something like:
```yaml
Cabin.Door:
  type: branch
  instances: InstanceLabels.InCabinZone
```

```yaml
Seat:
  type: branch
  instances: InstanceLabels.InCabinZone
```

Then, I would maintain separately this `InCabinZone` enum
```yaml
InstanceLabels.InCabinZone:
  - OPEN
  - OPENING
  - CLOSED
  - CLOSING
```

erikbosch, 2025-01-28:
MoM:

- Issue presented at meeting
- May be discussed in a subsequent meeting

UlfBj, 2025-02-03:
I like the idea of defining reusable data in the Types tree.

For reusable nodes I think the proposed syntax is good.

I added a Description to the Types node to stay conformant with the VSS rules.
 ```
Navigation.DestinationSet.Latitude:
  datatype: Types.LatitudeDegrees
  type: actuator
  description: Latitude of destination in WGS 84 geodetic coordinates.

Types.LatitudeDegrees
  type: datatype
  datatype: double
  min: -90
  max: 90
  unit: degrees
  description: Template for nodes defining degrees of latitude
 ```

For allowed values the legacy syntax example below:
 ```
Vehicle.Cabin.Convertible.Status:
  datatype: string
  type: sensor
  allowed: ['UNDEFINED', 'CLOSED', 'OPEN', 'CLOSING', 'OPENING', 'STALLED']
  description: Roof status on convertible vehicles.
 ```

Could be replaced by the following:
 ```
Vehicle.Cabin.Convertible.Status:
  datatype: Types.OpenState
  type: sensor
  description: Roof status on convertible vehicles.

Types.OpenState:
  type: allowedtype
  datatype: string
  description: Template for allowed values of openable sensors.
  allowed: ['UNDEFINED', 'CLOSED', 'OPEN', 'CLOSING', 'OPENING', 'STALLED']
 ```
The idea with allowed definitions in the Types tree is used in the CVIS project, so this would align the projects on this.

erikbosch, 2025-02-04:
MoM: 

- Ulf presented how it is used in CVI tree, but syntax proposed there is not necessarily same that is used it.
- We have a tool (him configurator) that runs before vss-tools to convert to format supported by vss-tools

erikbosch, 2025-02-05:
So it seems no-one objects extending the VSS type concept to support definition of reusable enum-types and restricted numerical types. There are however two aspects to consider.

- Does anyone have time and interest to define syntax in detail and to add support in vss-tools?
- How do we want to handle them in vss-tools output? I see two alternatives. One is to just "expand" the definitions so that output will look just like today. I.e. basically treating the enum typedefs as some form of macro. The other alternative is to keep them in the types file (or similar) just like we handle structs today, but that may gives problems for some downstream implementations that does not support structs at all today. This could possibly be solved by a flag in vss-tools.

UlfBj, 2025-02-05:
I think it should be possible to set a flag so that the tool "expands" the reference. Another reason to support this is efficiency when a server needs to access this data, so that it does not need to access the reference by also traversing the Type tree.

jdacoello, 2025-02-12:
If we agree on adding this to the modeling language, then I'd rather keep it simple. For example, by defining the enums in a separate place. I would not mix them with the structs. Such as in this example https://github.com/COVESA/vehicle_signal_specification/issues/802#issuecomment-2612581097 . Instead of re inventing them, we should at least adopt the ideas and patterns from other established languages https://github.com/COVESA/vehicle_signal_specification/issues/802#issuecomment-2609526693 .