COVESA/vehicle_signal_specification #804: Instances in vehicle_signal_specification/spec/Vehicle/MotionManagement /Steering.vspec for Axle are probably wrong

Herrmann-Alexander, 2025-01-28:
[image]
While looking trough the VSS signals I saw that the instances for Axle in vehicle_signal_specification/spec/Vehicle/MotionManagement /Steering.vspec were Row[1,1] while they should probably be Row[1,2]

erikbosch, 2025-01-28:
When proposing the model we opted for having `Row[1,1]` as default number of instances, as relatively few vehicles have four wheel steering, i.e. better to have just one instance as default, so that users can change to two instances in an overlay (or in a fork) if they need support for four wheel steering.

Do you have a need for four wheel steering in your VSS usage?

Herrmann-Alexander, 2025-01-28:
No, I don't have a need for four wheel steering. I just thought it might be unconsciously and wanted to point it out if it's a typo or something. But since it's intended I can close

erikbosch, 2025-01-28:
But we should at least add a comment/rationale, I can create a PR for that and bring up the discussion

jdacoello, 2025-02-12:
Isn't it the same as:
```yaml
...
  instances: [Row1]
...
```

I think the confusion is caused by the use of the notation `PREFIX[start,end]` which is intended to expand all the possible combinations.
Here, if the range will expand into only 1 label, then using simple `Row1` as instance label would do the job.

erikbosch, 2025-02-12:
Yes, but then we would use a different pattern compared to other `Row` instantiations. But i have no strong opinion if we rather would like to use/recommend `instances: ["Row1"]` if there is only one row and  `instances: Row[1,n]` if there are more. You can discuss in next meeting.

PR #809 (2025-02-27): Improve instance documentation
Two parts

- Explain why we currently have Row[1,1] for steerable axle
- Add some general info on the design decisions that we (implicitly) have made over time for instances

Fixes #804
commit 5b2efd6c (2025-02-27): Improve instance documentation (#809)

* Improve instance documentation

Two parts

- Explain why we currently have Row[1,1] for steerable axle
- Add some general info on the design decisions that we (implicitly) have made over time for instances

Fixes #804
closed 2025-02-27 by erikbosch