COVESA/vehicle_signal_specification #810: Duplicated name: Axle.Wheel

jdacoello, 2025-02-06:
I will try to implement it then.

BTW, it seems that while this fix was being discussed, the concept `Vehicle.MotionManagement.Brake.Axle.Wheel` was introduced to the spec. Which is again creating the undesired duplicated names:
```
                                            
fqn                                                  
Vehicle.Chassis.Axle.Wheel                 Axle.Wheel
Vehicle.MotionManagement.Brake.Axle.Wheel  Axle.Wheel
```

I will create another issue and suggest the fix for it.

_Originally posted by @jdacoello in https://github.com/COVESA/vehicle_signal_specification/issues/799#issuecomment-2639374645_

jdacoello, 2025-02-06:
Inside `Vehicle.MotionManagement.Brake.Axle.Wheel`, we have:
```yaml
Torque:
  datatype: int16
  max: 0
  unit: Nm
  type: sensor
  description: Estimated friction brake torque. Brake Torque < 0Nm.

TorqueArbitrated:
  datatype: int16
  max: 0
  unit: Nm
  type: sensor
  description: Brake system internally calculated friction brake torque target. Brake Torque < 0Nm.

TorqueFrictionMaximum:
  type: actuator
  datatype: int16
  max: 0
  unit: Nm
  description: Maximum wheel torque request for friction brake. Brake Torque < 0Nm.

TorqueFrictionMinimum:
  type: actuator
  datatype: int16
  max: 0
  unit: Nm
  description: Minimum wheel torque request for friction brake. Brake Torque < 0Nm.

OmegaUpper:
  type: actuator
  datatype: uint16
  unit: rad/s
  description: Upper wheel speed limit request controlled by friction brake. ISO 8855 wheel-spin velocity.

OmegaLower:
  type: actuator
  datatype: uint16
  unit: rad/s
  description: Lower wheel speed limit request controlled by friction brake. ISO 8855 wheel-spin velocity.
```

Inside `Vehicle.Chassis.Axle.Wheel` we have:
```yaml
Brake:
  type: branch
  description: Brake signals for wheel

Brake.FluidLevel:
  datatype: uint8
  type: sensor
  unit: percent
  max: 100
  description: Brake fluid level as percent. 0 = Empty. 100 = Full.

Brake.IsFluidLevelLow:
  datatype: boolean
  type: sensor
  description: Brake fluid level status. True = Brake fluid level low. False = Brake fluid level OK.

Brake.PadWear:
  datatype: uint8
  type: sensor
  unit: percent
  max: 100
  description: Brake pad wear as percent. 0 = No Wear. 100 = Worn.

Brake.IsBrakesWorn:
  datatype: boolean
  type: sensor
  description: Brake pad wear status. True = Worn. False = Not Worn.

#
# Tire
#
Tire:
  type: branch
  description: Tire signals for wheel.

Tire.Pressure:
  datatype: uint16
  type: sensor
  unit: kPa
  description: Tire pressure in kilo-Pascal.

Tire.IsPressureLow:
  datatype: boolean
  type: sensor
  description: Tire Pressure Status. True = Low tire pressure. False = Good tire pressure.

Tire.Temperature:
  datatype: float
  type: sensor
  unit: celsius
  description: Tire temperature in Celsius.

#
# Wheel signals
#

Speed:
  datatype: float
  type: sensor
  unit: km/h
  description: Linear speed of a vehicle's wheel.

AngularSpeed:
  datatype: float
  type: sensor
  unit: degrees/s
  description: Angular (Rotational) speed of a vehicle's wheel.
  comment: Positive if wheel is trying to drive vehicle forward.
           Negative if wheel is trying to drive vehicle backward.
```

erikbosch, 2025-02-06:
The challenge as I see it with these signals is that both `Axle` and `Wheel` are instances, so if we are to rename them we need to select something that gives a good meaning, and and the same time avoid to get too long expanded paths. We could for instance say that the branches within MotionManagement concerns dynamics only, and thus rename the branches accordingly. But doing that for all non-unique branches could also for example result in that `Vehicle.MotionManagement.Brake.Axle.Row1.Wheel.Left.Torque` would be replaced by `Vehicle.MotionManagement.BrakeDynamics.AxleDynamics.Row1.WheelDynamics.Left.Torque` which does not really make the name easier to understand. Even if we just would change it on for example axle level it would still result in `Vehicle.MotionManagement.Brake.AxleDynamics.Row1.Wheel.Left.Torque`.

In VSS as of today we do no really separate between "class-name" and "instance-name". Now and then I have thought if we possibly should support another attribute `typename` to be able to use the short name (i.e. `brake`) in getters/setters or a part of the fully qualified name, and type-name when generating for example classes/structs in the target format/language.

```
Brake:
  typename: MotionManagementBrakeInformation
  type: branch
  description: Brake signals for wheel
```

jdacoello, 2025-02-06:
In my opinion, there must exist only one `Wheel` with all the possible elements linked to it (i.e., either a leaf or another branch). For example:

```yaml
# Those from Vehicle.MotionManagement.Brake.Axle.Wheel
Wheel.Torque
Wheel.TorqueArbitrated
Wheel.TorqueFrictionMaximum
Wheel.TorqueFrictionMinimum
Wheel.OmegaUpper
Wheel.OmegaLower

# Those from Vehicle.Chassis.Axle.Wheel
Wheel.Speed
Wheel.AngularSpeed
Wheel.Brake. # pointer to other branch
Wheel.Tire. # pointer to other branch
```

This simple pattern will decouple the maintenance of the others, like:
```yaml
Brake.FluidLevel.
Brake.IsFluidLevelLow
Brake.PadWear
Brake.IsBrakesWorn

Tire.Pressure
Tire.IsPressureLow
Tire.Temperature
```
> Similar duality must be resolves for concepts like `Brake` and `Tire`. 

It seems that what is missing is a way to label a particular collection of properties. 
To keep it simple, I see two options:

## Option 1: Add another field `collection` in the specification of leaves:

```yaml
UniqueBranch.Property1
UniqueBranch.Property2
UniqueBranch.Property...
UniqueBranch.PropertyN
```

Where a collection can be optionally specified in any of those properties. For example, let's say you specify the collection label `MotionManagement` for properties :
UniqueBranch.Property1:
  type: attribute
  collection: MotionManagement
  ...

UniqueBranch.PropertyN:
  type: attribute
  collection: MotionManagement
  ...

 ## Option 2: Maintain the collection in another file with pointers to the concepts you are grouping.
```yaml
MotionManagementCollection:
  - UniqueBranch.Property1
  - UniqueBranch.PropertyN
```

Such that, you end up with a collection of only a subset of the specified properties for a UNIQUE branch!

erikbosch, 2025-02-11:
MoM:

- Presented in meeting
- Please review
- Sebastian: Do we want to switch to extend to more a powerful model, not necessarily good idea to make it more complex
- Paul: VSS would be a simple view of view of the model, there might be other more expressive views
- Sebastian: We need to be careful so that the "complex model" can be represented as a "good-looking" "classical" VSS-model.
- Ulf: VSS can exist without knowing anything about relationships.
- Adnan: Do we mean the same thing when talking about relationships, we need examples. 
- Sebastian: You should not need to be a data scientist to add items.

jdacoello, 2025-02-12:
@erikbosch Was there any concrete proposal on how to fix this duplication of branches (i.e., the issue itself)?

> Sebastian: Do we want to switch to extend to more a powerful model, not necessarily good idea to make it more complex

I guess the comment came from @SebastianSchildt:
I agree that it is not a good idea to continue to increase the feature set of the current vss language (i.e., customised yaml) The main reason is the required development and maintenance of the tooling. I support the idea of changing the language to something that is already established in communities outside COVESA and that supports all the current features of VSS (plus a few others). For example, the [Schema Definition Language (SDL)](https://graphql.org/learn/schema/) is a good candidate (simple, compact, easy to read). 

> Sebastian: We need to be careful so that the "complex model" can be represented as a "good-looking" "classical" VSS-model.

I agree. A yaml export could be provided.

> Ulf: VSS can exist without knowing anything about relationships.

@UlfBj It absolutely can, for sure. If you don't need cross references at all, and if you don't mind having an arbitrary one-dimension classification hierarchy (even for hundreds of concepts), then you would be well served with the current state (which is also fine). We are discussing here ways to go beyond what we have and to improve the modeling with the adoption of best practices. I believe that any change must guarantee backward compatibility as much as possible.

> Sebastian: You should not need to be a data scientist to add items.
I agree. Contributing to the controlled vocabulary should be as easy as possible. Improving the modeling approach must not compromise the simplicity.

UlfBj, 2025-02-12:
Daniel: It absolutely can, for sure. If you don't need cross references at all, and if you don't mind having an arbitrary one-dimension classification hierarchy (even for hundreds of concepts), then you would be well served with the current state (which is also fine).

@jdacoello  All of that can of course be done in a layer on top of VSS, without VSS having to know anything about it. Below VSS there is the native layer (CAN, etc). It does not have to know anything about these relationships. So the same can be true for VSS. Please let me know why that should not be possible.

jdacoello, 2025-02-12:
> All of that can of course be done in a layer on top of VSS, without VSS having to know anything about it. Below VSS there is the native layer (CAN, etc). It does not have to know anything about these relationships. So the same can be true for VSS. Please let me know why that should not be possible.

@UlfBj Everything is possible. However, the question is at what cost? 
If the vspec modeling language would stay as is, then you can build any layer on top. However, we see that the modeling language keeps changing over time. For example, by adding new constructs like `structs`, possible `profiles`, work arounds for the `units` that were not properly modelled several years ago, multiple trees, etc.

In my view, these changes occur because of the lack of clarity in the feature set. Today, we see the importance of new features that were not part of the initial design of the vspec language. 

To make an analogy, I consider the modeling language as the pilars for building a house. It is not possible to build a robust house if the Pilars are being modified while building the house.

To be more concrete, I here is a list of the requirements that I IMHO the current modeling language does not satisfy:

[image]

UlfBj, 2025-02-12:
@jdacoello I would say that layered architectures typically show lower cost over time than more monolithic architectures.
The lacking features in your list can be supported in a layer on top of VSS.

SebastianSchildt, 2025-02-12:
If I could wish for something, is the following possible:

**Assumption:**
We switch all our current YAML files to a new technology `X` offering everything we desire. We immediately make use of the features, that allows us to define and reuse "concepts" (as opposed to the current mor macro-like "repeat stuff" feature).

**Is this possible:**
 - Within this "new" format I can define an entrypont "Vehicle" that references everything so that
    - a tree (maybe flat vspec.yaml) can generated, that looks (largely) the same as a current VSS tree runs through our current tooling, so that @UlfBj  (also me, KUKSA, VISS, current implementers) would never notice
    - we could - reusing the same basic concepts - also define lists/trees for use cases such as "Commercial Vehicles" or for domains (such as Vehicle Motion) - because that would be an additional value. That would actually go in the same direction what HIM also tries to achieve (whether this is the "better way" or just a good  way to generate HIM trees is probably another discussion based on effort of tooling)

**Potential Risk**
If now we work with our newfound powers for a couple of months, are we sure, this still works? I mean, of course fancy relations and reusing of concepts are only fully expressible in the "new format", but can we - to take the dumbest example we have - still generate a CSV output that contains a meaningful representation of "all that is defined in the standard catalogue"?
My potential fear is, ending up in a situation where suddenly we say "largely we can convert everything the classic view, but not wheels because of fancy feature x,y,z". THAT @UlfBj  (also me, KUKSA, VISS, current implementers) would notice. 
And from my point of view/work **in vehicle**  I see VSS as sort of "fancy SDV signals", so a bit more clever than CAN but not too much (like Ulf said, not so much relations on this level). I want to work with Key/Identifier->value maps and not be required to put any heavyweight (semantic/reasoning) stack OR basically invent something like "old VSS" as mapping layer myself.

Is this fear unfounded, or is there anything we can do/define/code to prevent that @jdacoello ?

If this is possible and I can be convinced the risk is manageable, I for one would like to get rid of a lot of vss-tools plumbing maintenance in the long run, while winning some features enabling more use cases.

SebastianSchildt, 2025-02-12:
> [@jdacoello](https://github.com/jdacoello) I would say that layered architectures typically show lower cost over time than more monolithic architectures. The lacking features in your list can be supported in a layer on top of VSS.

Generally I agree, but I think for some of this features it is harder than it should be. Because you would - manually - at least supervised - teach the upper layer system that "all wheels are actually the same concept", or "seats are, except that seat", that is not easy to do fully automated.  Same if you want to express lower level things like "a temperature always looks like this". I feel on a slightly related topic for the units/quantitites we also did already a bit or "reinventing" and "roll our own impl" in vss-tools compared to some other things that may already be there.

So if there is - an easy "I am not a data scientiest" - kind of way to put such information in our VSS files, where you can *decide* whether  to use or ignore  this information in generators (similar as today there are systems that do not care mich for description or even unit), that would be a win. Because it would make building that "next layer" on top of VSS easier

UlfBj, 2025-02-12:
>> "all wheels are actually the same concept", or "seats are, except that seat",

@SebastianSchildt the problems you mention are solved in a VSS compatible way in the CVIS project.
The VSS project is very welcome to adopt the solutions.

jdacoello, 2025-02-12:
@SebastianSchildt I am selecting some concepts from VSS to show a concrete example for an alternative language. Please, just let me know which concepts from the current spec you think might be problematic and I can showcase them. Then, we will have a clearer answer to your concern. At first sight, I would say that need to agree on how the extra features that are not part of the current vspec language have to be represented.

> layered architectures typically show lower cost over time than more monolithic architectures.
@UlfBj I think the modeling language is one building block only, which must be stable to be used. If we have different layers just for the language, then maintaining that is harder.

jdacoello, 2025-02-12:
> VSS is used as is in many contexts, where e. g. VISS is one.

What is the dependency in that case? Is it the `path` that is used as the identifier?  I think I need concrete cases to test and show we're not braking anything. 
I suggest that we start documenting the concerns in this discussion https://github.com/COVESA/vehicle_signal_specification/discussions/813 . This way, we can make sure we're not overlooking something important.

erikbosch, 2025-02-12:
> [@erikbosch](https://github.com/erikbosch) Was there any concrete proposal on how to fix this duplication of branches (i.e., the issue itself)?
> 

Nope.

Looking at how we use branch-names today they are quite often used just to group things (like having `Brake.PedalPosition` rather than `BrakePedalPosition`) or to handle instantiation. Where to put something logically is not always straightforward. As an example, we have `Vehicle.Cabin.HVAC.IsRearDefrosterActive` but it can be discussed if the defroster "belongs" to the HVAC system or if it rather should be a property of `Vehicle.Body.Windshield.Front`. I do not really have a strong opinion.

Concerning actual names for this particular case I think we first need to agree on if `Brake` is an "allowed" name. So far we have used "Brake" here and there for items that in one way or another are related to the brake system. See some examples below. Even if we for now only consider "two-segment branches" like `Axle.Wheel`we should better consider also one segment names when renaming them. So we should better align on which level we want to have a branch `Brake`. Vehicle-level? Axle-level? Wheel-level?

```
Vehicle.Chassis.Brake.PedalPosition
Vehicle.Body.Lights.Brake.IsActive
Vehicle.Chassis.Axle.Row1.Wheel.Left.Brake.FluidLevel
Vehicle.MotionManagement.Brake.VehicleForceMaximum
Vehicle.MotionManagement.Brake.Axle.Row1.TorqueElectricMinimum
Vehicle.MotionManagement.Brake.Axle.Row1.Wheel.Right.TorqueFrictionMaximum
```

One possible (long term) layout could be to use the term `Brake` only as branch on top level and move all vehicle-level brake-related signals there, like:

```
# Vehicle wide data grouped around feature "Brake"
Vehicle.Brake.PedalPosition 
Vehicle.Brake.LightIsActive
Vehicle.Brake.VehicleForceMaximum
```
or if we think `Vehicle.Chassis` is good happy home

 ```
# Vehicle wide data grouped around feature "Brake"
Vehicle.Chassis.BrakePedalPosition 
Vehicle.Chassis.BrakeLightIsActive
Vehicle.Chassis.BrakeVehicleForceMaximum
```

Then on other levels we would only use `Brake` as prefix on indiviual signals

```

# Axle level signals, use prefix Brake to indicate that signals are brake-related
Vehicle.Chassis.Axle.Row1.BrakeTorqueElectricMinimum

# Wheel level signals, use prefix Brake to indicate that signals are brake-related
Vehicle.Chassis.Axle.Row1.Wheel.Left.BrakeFluidLevel
Vehicle.Chassis.Axle.Row1.Wheel.Left.BrakeTorqueFrictionMaximum
```

That mean in practice moving the Vehicle Motion signals to "Wheel", "Axle" and "Vehicle.Brake" (or "Vehicle.Chassi"). That would work if we want to go in that direction.

But then I think that we at least in vspec source format should have a mechanism to specify which feature a specific signal belongs to, like

`feature: motion_management`

or `doc: https://covesa.github.io/vehicle_signal_specification/catalog/vehicle_motion_management/index.html`

The reason is that so far we have to a great extent documented design rationale in source files, like in https://github.com/COVESA/vehicle_signal_specification/blob/master/spec/include/PowerOptimize.vspec . If we spread out signals in the tree we need a better way for documenting the intended use of a specific signal, with a tag or a link that can help you to find the corresponding section in documentation.