COVESA/vehicle_signal_specification #815: Release Planning 2025

erikbosch, 2025-02-25:
- Shall we release 5.1 or 6.0
- At next AMM (May?)
- VSS-tools can be released separately
- What is NOT backward compatible (if any)

paulboyes, 2025-06-13:
I think we should consider a spring/summer release.

erikbosch, 2025-06-18:
Potential backward-incompatible changes (if so not to include in 5.1), to be discussed

- https://github.com/COVESA/vehicle_signal_specification/pull/793 (based on #780)
   - Adding instances to ObstacleDetection branch

erikbosch, 2025-06-24:
Planned content if to be created today (for VSS)

## What's Changed
* Prepare for next major release by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/783
* Separate doc-gen from general build by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/784
* Add Media.Played.Genre by @cws55 in https://github.com/COVESA/vehicle_signal_specification/pull/781
* Add RegExp pattern for VehicleIdentification.VIN property. by @Kostadin-Ivanov in https://github.com/COVESA/vehicle_signal_specification/pull/786
* Doc Update by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/788
* Add ConsumptionSinceLastRefuel by @CoffeeMoneyMakers in https://github.com/COVESA/vehicle_signal_specification/pull/787
* Add signals for Motion Management by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/785
* Doc Update by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/792
* Pr780 by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/793
* Fix 24.04 build by pinning Python version by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/794
* D3 now bypasses Observable Notebook framework by @Giaccomole in https://github.com/COVESA/vehicle_signal_specification/pull/801
* Clarify VSS expectations on read/write by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/796
* refactor: Unique names for the pattern {parent_banch_name}.{branch_name} by @jdacoello in https://github.com/COVESA/vehicle_signal_specification/pull/799
* Update pre-commit version by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/803
* Correction of typo in spec/Body/WiperSystem.vspec by @Herrmann-Alexander in https://github.com/COVESA/vehicle_signal_specification/pull/807
* Missing "s" in vehicle_signal_specification/docs-gen/content/rule_set… by @Herrmann-Alexander in https://github.com/COVESA/vehicle_signal_specification/pull/811
* Improve instance documentation by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/809
* VSS Extension Profile Documentation by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/817
* VISS documentation update by @UlfBj in https://github.com/COVESA/vehicle_signal_specification/pull/823
* Fix typo in build check by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/825
* Removing some recently added motion management signals. by @erikbosch in https://github.com/COVESA/vehicle_signal_specification/pull/831

erikbosch, 2025-07-14:
We have some definitions i https://covesa.github.io/vehicle_signal_specification/governance/index.html

In short - breaking changes should typically occur only in major releases, indicated by changing the major number. But the devil lies in the details, strictly speaking there are a few changes in the current master branch that can be considered as major:

- GraphQL generator has been refactored, so output is significantly different compared to vss-tools 5.0
- ObstacleDetection instances have refactored instance representation
- The VIN signal has an added pattern. This means that VSS "master" cannot be used used with vss-tools 5.0 in strict mode unless you add an argument to accept the pattern extended attribute. (If you would use vss-tools from "master" there is no problem)

Strictly speaking they are all major changes, so the interesting question here is if we think there are reasonable reasons to anyway treat them as "minor" changes, which could be the following assumptions:

- No one uses the "old" GraphQL tool
- No one uses the current obstacle signals
- We do not consider VSS changes backward incompatible if they can be "solved" by using the latest "matching" minor release of vss-tools.

So maybe it is good to refine when/how/if it is OK to treat a major change as minor.

Kostadin-Ivanov, 2025-07-15:
Hello Erik (@erikbosch),
Good observation :) 
I agree with you and personally, I think that all of the listed updates might push VSS to 6.0.0

Reason is that we have an "exporter" updates + VSS structure updates for the obstacle signals and VIN - regExp validation pattern.

With regards to VIN - pattern, as far as I remember, there was some updates in the VSS-Tools so to handle the NEW: **pattern** signal attribute.

Thus, for consistency I'd say that it would depend on both VSS and VSS-Tools versions. If VSS-Tools goes with a major version increase due to some updates requested to handle changes in VSS, then VSS would also have to go 1 major version increase and vice versa.

Just a side question.
Since we currently are on VSS: 5.0, then the new version would be either: 5.1 or 6.0, depending on which versions increment we will pick, right?

erikbosch, 2025-07-15:
If we decide that some items to be considered major, then the workflow would likely be to just cherry-pick those items that not are major to release a minor version. Previously we have said that we do not want to release major version too often, as we want major version to also include significant improvements, not only a type/unit change for a specific signal or similar.

Kostadin-Ivanov, 2025-07-17:
> If we decide that some items to be considered major, then the workflow would likely be to just cherry-pick those items that not are major to release a minor version. Previously we have said that we [...]

Then, I guess the answer on the whether to go major or minor is very easy :)

Happy VSS 5.1 :)