COVESA/vehicle_signal_specification #818: EU Battery Passport and VSS

erikbosch, 2025-03-27:
This ticket is a placeholder for discussions/plans related to the EU Battery Passport. Topics to discuss:

- How does it relate to VSS
- Any need to review or extend VSS data to cover the battery passport data
- Any interest in that

-     https://thebatterypass.eu/
-     https://www.dinmedia.de/en/technical-rule/din-dke-spec-99100/385692321

SebastianSchildt, 2025-04-01:
There seems to be "THE" (an?) implementation https://github.com/batterypass/BatteryPassDataModel

Discussion from meeting 2025/04/01: preferably if this feature is wanted in VSS instead of "copying the DIN once again", maybe the model can be imported. 

A bit unclear whether there is tooling or it needs "reversing" ttls...

Related: https://github.com/batterypass/BatteryPassDataModel/issues/12

nd-circulor, 2025-04-01:
The referenced battery pass data model above is created with the same tooling that is used at CATENA-X. It's the [Eclipse Semantic Modeling Framework](https://eclipse-esmf.github.io/esmf-documentation/index.html) with the modelling tool is the [Aspect Model Editor](https://eclipse-esmf.github.io/ame-guide/introduction.html) which is mentioned in the Readme file. The Github repository structure complies with the Aspect Model Editor and the corresponding namespace. There's currently a joint working group between CATENA-X and IDTA to harmonise the battery pass model with the existing IDTA ASS sub models like digital name plate, technical information, etc.

erikbosch, 2025-04-02:
As ESMF was mentioned in last comment - it can be mentioned that ESMF is not new to VSS, we have some support in the other direction described at https://github.com/COVESA/vss-tools/blob/master/docs/samm.md