COVESA/vehicle_signal_specification #822: DistanceTraveled granularity

sophokles73, 2025-05-05:
The *Vehicle.DistanceTraveled* is currently measured in kilometers. Is there a particular reason why it is not measured in meters? The Fleet Management Standard requires the total distance traveled to be tracked at meter granularity. J1939 includes SPN917 (Total Vehicle Distance (High Resolution)) for this purpose and it would be nice if we were able to map this into VSS without losing precision.

erikbosch, 2025-05-06:
The signal has been there for a long time and I assume that `km` was selected as that is the typical unit used when describing a vehicle in a used car ad or when displaying the value to the driver. It is an VSS float value (IEEE 754-2008 binary32 floating point number) so if using the default VSS type it might not be able to keep meter precision if the vehicle has traveled a long distance.

```
TraveledDistance:
  datatype: float
  type: sensor
  unit: km
  description: Odometer reading, total distance traveled during the lifetime of the vehicle.
```

We have in general historically been positive to changing datatype/units when needed. One option would be to change to meters and change datatype to uint32 (or uint64 if we think uint32 does not suffice). Another option would be to change datatype to double, that would give sufficient significant digits to support meter precision regardless of whether using meter or kilometer as unit.

erikbosch, 2025-05-06:
MoM:

- presented at weekly meeting.

SebastianSchildt, 2025-05-06:
To see what precision you have for larger values with float/double this is helpful: https://float.exposed/

sophokles73, 2025-05-06:
Having played around a little with the link provided by @SebastianSchildt I guess that I would be in favor of changing the unit to meters and using a uint32 instead of using a float/double.
A uint32 would support distances up to 4.29 million kilometers. That should be sufficient even for commercial vehicles which tend to be on the road longer than the average passenger car, shouldn't it?

erikbosch, 2025-05-08:
Feel free to open a PR, then people have to say yes/no.

erikbosch, 2025-12-02:
MoM:

- Brought up by Ted, increased granularity needed for particular signals
- There are two proposed changes, traveled distance (#824) and #827 (engine speed)
-