COVESA/vehicle_signal_specification #833: Pattern keyword is 6.0 or 5.1?

SebastianSchildt, 2025-07-01:
In 5.1 release candidate mentions support for "pattern"

https://github.com/COVESA/vehicle_signal_specification/releases/tag/v5.1rc0

is this really in the vspec already or is it a tooling thing? (then shoudl not be mentioned?) , The documentation for pattern here https://covesa.github.io/vehicle_signal_specification/rule_set/data_entry/sensor_actuator/index.html mentions it is a "since 6.0"

This should be consistent. If "pattern" does not break anything, no problem to add it in 5.1

erikbosch, 2025-07-16:
After the discussion at the VSS-meeting yesterday were most participants seem to prefer a strict approach I have created a 5.1rc1 release that does not include the "pattern" change. The reason is that it can be seen as a backward incompatible change, if running in strict mode you will get an error if using vss-tools 5.0. I also suggest not adding the pattern functionality to vss-tools 5.1 as the master branch of vss-tools contain other backward incompatible changes, adn cherry-picking only certain tickets would be quite cumbersome. So I suggest not releasing vss-tools 5.1, and instead let vss 5.1 still use vss-tools 5.0

Related to #815

erikbosch, 2025-08-05:
Was not included in 5.1 so discussion no longer relevant

closed 2025-08-05 by erikbosch