COVESA/vehicle_signal_specification #836: Differentiation between TPMS and TMS sensors value

JMorceaux, 2025-07-17:
Hello VSS Team !

When looking at the signal `Vehicle.Chassis.Axle.Row1.Wheel.Left.Tire.Temperature`, one could ask about the nature of the temperature: are we measuring the temperature of the rubber, or the air inside the tire.

With my team, we deal with cars equipped with TPMS or TMS and that would be a nice addition to differentiate if the temperature returned by the sensors is the temperatore of  the air or the rubber.

What we are proposing is:

- Proposition n°1:
    - We replace the field `Temperature` with `TireTemperature` that would describe the temperature of the rubber and add a new field `AirTemperature` that would describe the temperature of the air inside the tire.

```
Tire.TireTemperature:
  datatype: float
  type: sensor
  unit: celsius
  description: Rubber temperature of the tire in Celsius.

Tire.AirTemperature:
  datatype: float
  type: sensor
  unit: celsius
  description: Air temperature inside the tire in Celsius.
```

- Proposition n° 2:
    - We keep the field `Temperature`, just change the description and add the new field `AirTemperature`

```
Tire.Temperature:
  datatype: float
  type: sensor
  unit: celsius
  description: Rubber temperature of the tire in Celsius.

Tire.AirTemperature:
  datatype: float
  type: sensor
  unit: celsius
  description: Air temperature inside the tire in Celsius.
```

As I understood, the guidelines would point out more towards the second proposition. Lemme what you think about this change
Cheers,

Jérémy

erikbosch, 2025-07-22:
MoM:

- Presented at meeting.
- Please discuss
- E: I could guide/help with creating a PR to discuss

JMorceaux, 2025-07-24:
Hello @erikbosch !

Sorry for the late reply, I've been trying to gather information internally as much as possible to refine my answer. It's also a challenge to filter out what can be said and what cannot :)

If we look at both sensor, here are the main differences:

|                    | TPMS                                   | TMS                                                | 
|--------------------|----------------------------------------|----------------------------------------------------|
| Mount position | Tire valve (most cases)                | On the inner surface of the tire |
|                    | Around the rim (especially for Trucks) |                                                    |   |   |
| Temperature Sensor | Air temperature                        | Surface temperature of the rubber                 |

For TMS, the temperature sensor is in contact with the rubber whereas the TPMS is only in contact with the air inside the tire. All I can say is that the value can vary by a lot, especially in performance mode.

Rubber is a very bad thermal conductor ([Rubber thermal properties](https://www.iltusa.com/thermal-properties-of-rubber/)) so to answer your question about measuring the temperature: it's not easy, you do not have a single value for the entire tire. So the TMS temperature reading will only give information about the temperature of the inner surface of the tire but still is better reading than what TPMS can give.

That is why, we believe creating a second temperature field for TMS sensor would be appropriate.

commit 1624e7a9 (2025-08-19): Extend tire temperature signaks

Fixes #836
closed 2025-08-19 by SebastianSchildt