COVESA/vehicle_signal_specification #846: Clarification on datatype choices for percentage signals (float vs uint8)

yosuke-lab, 2025-09-05:
Hello,

while learning this repository I noticed that signals expressed in percent sometimes use different datatypes.

For example:

## Backrest.Lumbar.Support
[link](https://github.com/COVESA/vehicle_signal_specification/blob/1624e7a95e574b4618222e7d4bc3d68e7cee4112/spec/Cabin/Seat.vspec#L115)

```
Backrest.Lumbar.Support:
  datatype: float
  type: actuator
  unit: percent
  min: 0
  max: 100
  description: Lumbar support (in/out position). 0 = Innermost position. 100 = Outermost position.
  deprecation: v5.0 - refactored to Seat.Backrest.LumbarSupport
```

## Position
[link](https://github.com/COVESA/vehicle_signal_specification/blob/1624e7a95e574b4618222e7d4bc3d68e7cee4112/spec/include/MovableItem.vspec#L27)

```
Position:
  datatype: uint8
  type: actuator
  min: 0
  max: 100
  unit: percent
  description: Item position. 0 = Start position 100 = End position.
  comment: Relationship between Open/Close and Start/End position is item dependent.
```

Both signals represent a percentage from 0 to 100, but one is defined as float and the other as uint8.
I would like to better understand the rationale behind this difference:

* Is this an intentional design choice (e.g., depending on expected precision)?
* Or is it more of a historical / consistency matter?

Is there any general guideline in VSS about which datatype should be used when defining percentages?
Thanks in advance for the clarification! 🙏

erikbosch, 2025-09-05:
Thanks for your interest. There are no strict rules, but historically COVESA/VSS has tried to select the smallest type possible that still fits the use-cases you want to support.

Taking fuel level, battery charge status or audio volume (in percent) - it is likely not necessary to represent the value with decimals as it typically is displayed without decimals. But for things that can be actuated like seat position and window position you may want to have higher resolution as the range of movement might be quite big and 1% might be a too big gap, and then sometimes float has been selected.

This might indeed lead to inconsistencies and if you think that 1.0% resolution isn't sufficient for a specific signal feel free to propose that the signal shall change to float.

As a side note - the use of float or uint8 in VSS does not need to imply that the same datatype is used internally in the vehicle. In e.g. CAN networks it is quite common to use a scaling to represent decimals/fraction, like using a uint8 but specifying that each bit means 0.4%.  In that case - even if for example fuel level is reported to backend as a float following VSS spec, actual value received from vehicle network will only have 0.4% resolution, like 0%, 0.4%, 0.8% and so on.

yosuke-lab, 2025-09-05:
@erikbosch 
Thank you for your detailed reply!  
I understand now that the datatype is chosen individually for each signal based on the required resolution and typical use cases.  

As you pointed out, it is also important to consider the relationship with the resolution available on the vehicle network (e.g. CAN scaling). I’ll take this into account and see how best to adapt it to the environment I’m working in.  

Thanks again for the clarification 🙇