COVESA/vehicle_signal_specification #848: Found problem with datatype in Vehicle.MotionManagement.Brake.Axle.TorqueElectricMinimum

Kostadin-Ivanov, 2025-09-18:
After I upgraded my local VSS to latest: 5.1. I tried to convert it to SAMM aspect models, using VSS-Tools SAMM exporter and got below error:

`
INFO     VSpecs loaded, amount=74                                                                                                                                                           vspec.py:130
CRITICAL 'Vehicle.MotionManagement.Brake.Axle.TorqueElectricMinimum' has 1 model error(s):                                                                                                   main.py:231
                    [
                        {
                            'type': 'assertion_error',
                            'loc': (),
                            **'msg': "Assertion failed, max '0.0' is not an 'int16'",**
                            'input': {
                                'fqn': 'Vehicle.MotionManagement.Brake.Axle.TorqueElectricMinimum',
                                'type': <NodeType.ACTUATOR: 'actuator'>,
                                'description': 'Limit for regenerative brake torque at given axle. Brake Torque < 0Nm.',
                                'comment': None,
                                'delete': False,
                                'deprecation': None,
                                'constUID': None,
                                'fka': [],
                                'instantiate': True,
                                'datatype': 'int16',
                                'max': 0.0,
                                'unit': 'Nm'
                            }
                        }
                    ]
`

It appears that the datatype was set to 'short' - int16, but the max value was given as float: 0.0.

sschleemilch, 2025-09-19:
I will have a look when I am back from vacation, which means possibly next week

sschleemilch, 2025-09-22:
This has been fixed on most recent VSS: https://github.com/COVESA/vehicle_signal_specification/commit/d1afee59cb74f4d3861eef37cbce093905f07b7c

But yeah, it is not yet released